PhotoboothProject/photobooth #10: [LOCALIZATION]:

andi34, 2022-08-27:
### Request a language

ad

### Additional context

ad

mgrl, 2022-10-03:
Hi, I made a 3d printed photobooth using a 16mp arducam autofocus camera.

## :camera: Photobooth in action
[image]

## :heart_eyes: Summary

- 3d printed design
- provided as is
- [solidworks maker design files available](https://github.com/mgrl/photobooth-3d/tree/main/cad)
- components listed in [BOM](https://github.com/mgrl/photobooth-3d/blob/main/docs/BOM.xlsx)
- 16MP autofocus camera (arducam)
- led ring signaling photo countdown and when the photo is actually taken
- 7" touchscreen
- for autofocus, led signaling and picture capture I use a small [imageserver setup](https://github.com/mgrl/photobooth-imageserver)

## :camera: Screenshots of the photobooth

[image]
[image]
[image]
[image]

obkram, 2023-01-05:
My photo box is a little older but it works perfectly.

[image]
[image]