PhotoboothProject/photobooth #12: [BUG]: Store Admin Panel Data

mgrl, 2022-08-28:
### Describe the bug

In latest dev version I experience a problem storing the config in the admin panel.
None of the changes will be written to the config file. After hitting "save" button on the admin panel, the page reloads and old values still persist.

### Steps to reproduce the behavior

Open Admin Panel
Change any value (example language or command to take a picture)
Click Save
Check values: Old values are loaded again.

### Expected behavior

Store the new values provided in textfields.

### Photobooth Version

dev from 2022-08-28

### Browser and browser version

firefox

### Operating System

raspi pi bullseye 32 bit

### Did it work before?

Yes

### Latest working version

not sure, think last dev from may was working

### Additional context

Using nginx as webserver.

mgrl, 2022-08-28:
Reverted via installer --update to branch package, which installed version tagged v3.3.0. That works okay again.

andi34, 2022-08-28:
How did you Update?

Please Run

```sh
sudo chown -R www-data:www-data /var/www/
```

And try again.

andi34, 2022-08-28:
If that also does not Work, please try deleting /renaming the existing config/my.config.inc.php.

andi34, 2022-08-28:
I am unable to reproduce. Please Post the Content of your config/my.config.inc.php so i can Check whats causing the issue (maybe an removed Option stored?).

mgrl, 2022-08-28:
Test later again. I think it might be something in the frontend because when clicking "Save" all the data is added as GET variables to the URL. In 3.3.0 it seems to be posted and the url does not get cluttered.

andi34, 2022-08-28:
Some Options have been removed, also some of them have been Hidden on the Frontend before. If one of those is stored inside the my.config.inc.php it _might_ cause problems.

mgrl, 2022-08-28:
It worked with another browser after update; so I cleaned the cache of the usually used browser again and it worked now :)  
So it seems, clearing the cache was essential.

closed 2022-08-28 by mgrl