PhotoboothProject/photobooth #29: [FEATURE]: easy access to private folder

andi34, 2022-09-13:
### Is your feature request related to a problem?

No

### Description

Currently it's quite hard for users to add own stuff inside the private folder because of limited permissions.

### Describe the solution you'd like

Change permissions of the private folder to make it easy for users to place own stuff.

### Describe alternatives you've considered

If the username is set while running the installer:
- add user to `www-data` group
- add a symlink inside the home directory of the user

Another option would be to add a own subpage with the possibility to upload images etc. To the private folder.

### Additional context

Should be discussed.

kreativmonkey, 2022-09-14:
What place does "private" folder mean? Is this the folder to place frames and fonds? I think the easiest way is to offer a upload and Selektion field on the settings page. The more often users do not need to log in with ssh or interact with the system, the more stable it will be  😉

andi34, 2022-09-14:
Yeah, it's for own fonts, backgrounds, frames and overrides.css which aren't tracked on GitHub (can't cause issues on Update because of uncommitted changed).

kreativmonkey, 2022-09-16:
Ok, thats like custom files in other projects. How you told, there are two different options. The easiest one for the programmer is to add the default user to the www-data group and set a sticky bit for the group in the private folder `chmod g+s /private`. So every file in the folder automatical get the group www-data and is readable by the webserver. 

Easier for the user are a upload page so they don't need to interact with the system behind photobooth itself. 

I think the best way is to start with the first one and afterwards develop a upload service in the admin settings page for the neccessary fields ore somthing else.

andi34, 2022-09-16:
So we would need to define a username always, currently it's only needed on a raspberry pi.
How about adding 766 perms to the folder instead? We wouldn't need to challenge with usernames in that case. With sticky bit it would be 7766.

andi34, 2022-09-16:
https://chmodcommand.com/chmod-7766/

up-87, 2022-09-16:
If we add a file upload we‘d need to limit it to some file types. Otherwise the file upload and the customizable scripts are an invitation to take over the system. Tbh I think an easier handling of the private folder would already be fine.

kreativmonkey, 2022-09-18:
for security reason it will be better to add a user for the application itself. The limitation of file types is good but it didn't prevent the system from take over. First of all the private folder should not be public access so you can't call a file directly so you can check files while import them and you have the control over what will happens ore not. 

@andi34 you don't need `7766` you only want `2766` becouse you only need to set gid flag so all files are in the `www-data` group.

andi34, 2022-09-21:
Since https://github.com/PhotoboothProject/photobooth/commit/31344cdbb66daa184680bccb8e141e74e793a0b4

We could do
```
chown $USERNAME:www-data
```

Instead of

```
chown www-data:www-data
```

kreativmonkey, 2022-09-28:
Yes you can do it that way ore you use acl to grand permission for the user.

andi34, 2022-11-27:
Keep open

andi34, 2023-01-26:
Still needed

closed 2023-04-17 by andi34