PhotoboothProject/photobooth #53: [FEATURE]:  remotebuzzer on live chroma

andi34, 2022-10-07:
### Is your feature request related to a problem?

No

### Description

Remotebuzzer Server is actually disabled on livechroma.

### Describe the solution you'd like

Enable support for livechroma.

### Describe alternatives you've considered

None.

andi34, 2022-12-06:
Keep open

andi34, 2023-04-18:
Still todo

flighter18, 2024-12-29:
@andi34 who is the progress of this feature? 
maybe I can help you?
I would love to take pictures in chroma by buzzer.

andi34, 2024-12-29:
@flighter18 might be it works already, can't say as I haven't used the keying for a long time now 😅
Restructuring and rewriting other sources had and have priority in the limited time I am able to spend to the project currently. Having 3 child's now which makes development more difficult to me.

andi34, 2024-12-30:
Right, it needs to trigger `photoBooth.thrill('photo');`. It's working while pressing the capture button at the point where trying the curl?
Anything visible inside the Remotebuzzer log?

closed 2025-03-21 by andi34