PhotoboothProject/photobooth #60: [BUG]: The "X" Button in Gallery - Pictureview is not clickable in the center

r0xx5, 2022-10-17:
### Describe the bug

Pressing the "Close" button directly in the center does not work. Only when you press outside the "X" it works.

### Steps to reproduce the behavior

1.home page
2.gallery button
3.open photo
4.try to close the photo view over the "X" in the top right corner.

### Expected behavior

Can Close the Pictureview within the X center.

### Photobooth Version

4.2.0

### Browser and browser version

Edge, Chrome (Raspi)

### Operating System

Windows 11, Raspian

### Did it work before?

No

mgrl, 2022-10-25:
I second this error. Is there any workaround avail?

andi34, 2022-10-25:
Not yet. I am not sure why it's happening, was unable to resolve this on photobooth source yet. Need to take a look at the PhotoSwipe source and check if it needs to be fixed there

closed 2022-10-28 by andi34

mgrl, 2022-10-28:
Works, thx!🥇