PhotoboothProject/photobooth #80: [BUG]: Live view not working

der1igedon, 2022-11-02:
### Describe the bug

I try to use the live view feature with my EOS 500D, but the picture won't light up in the backround. I'm using the newest version. I hear the mirror of the camera clicking, but it immidialty goes back and no live view is visible.

### Steps to reproduce the behavior

If I run `python3 cameracontrol.py --bsm` the following error shows up:

[image]

### Expected behavior

The live view should be visible in the background during picture taking.

### Photobooth Version

latest

### Did it work before?

No

andi34, 2022-11-03:
Hey and thanks for your report.
You don't have the GPhoto2 video Interface which will be used.
Have you installed the service as described in the FAQ?
If not, you should follow the steps mentioned in the FAQ.

If yes, you night news to fix the kernel module yourself.

### Fix v4l2loopback Module

Open a terminal and run the following commanda:

```sh
curl -LO https://github.com/umlaeute/v4l2loopback/archive/refs/tags/v0.12.5.tar.gz
tar xzf v0.12.5.tar.gz && cd v4l2loopback-0.12.5
make && sudo make install
sudo depmod -a
sudo modprobe v4l2loopback exclusive_caps=1 card_label="GPhoto2 Webcam"
```

Now check if everything is fine:
```sh
sudo v4l2-ctl --list-devices
```

If it didn't worked, you get the following response:

```
Cannot open device /dev/video0, exiting
```

If it worked, you get the following response:

```
GPhoto2 Webcam (platform:v4l2loopback-000):

        /dev/video0
```

closed 2022-11-03 by andi34

der1igedon, 2022-11-09:
Thanks a lot! This works.

andi34, 2022-11-09:
👍 great. I'll add a note to the update instructions asap. Maybe we should by default compile the module by ourselves.
But independent of the Photobooth Update: it always need to be re-done if the kernel gets updated on the system (regular System Update).

andi34, 2022-12-19:
https://github.com/PhotoboothProject/PhotoboothProject.github.io/commit/c3310f812799cc7dc1bdd40229d3923c539ac902