PhotoboothProject/photobooth #81: [BUG]: Language files not registered

mgrl, 2022-11-06:
### Describe the bug

Problem loading language files:
[image]

### Steps to reproduce the behavior

1 Fresh installation, new OS, starting with installation script
2 Photobooth was installed to subdirectory, so accessible by http://photobooth/photobooth
3 Find language files are always loaded from root directory instead subdirectory.

### Expected behavior

Load language files from subdirectory where installed by script.

### Photobooth Version

yesterday

### Did it work before?

Yes

andi34, 2022-11-06:
@mgrl the path is detected automatically, but it can be adjusted via adminpanel in case the detection of the path fails (e.g. because of wrong / special Webserver settings). Last time I've heard about that was because of a wrong defined document_root path which is used to detect the install path (`$_SERVER['DOCUMENT_ROOT'];`).

andi34, 2022-11-06:
[image]

mgrl, 2022-11-06:
I see. Good to know which setting to change. It's been complete fresh installation, latest RPI OS. Not sure why the autodetection might have failed.
Think it can be closed then.