PhotoboothProject/photobooth #82: [BUG]: After shot image is loaded 2 times

mgrl, 2022-11-06:
### Describe the bug

When a picture is captured, the image is loaded in different versions but actually only one is displayed.
I do not use any filters on the images.
After takepic.php is issued, three images are loaded (2 large versions, 1 thumbnail), see marked lines 
[image]

Now on raspberry pi it takes some seconds to display these large images (RPI 4). Anything I can do to optimize or avoid loading the filter jpeg also? I think in earlier versions of photobooth the image was displayed immediately, but not 100% sure when it got worse.

### Steps to reproduce the behavior

1 take picture on raspberry pi
2 image was captured
3 background livestream restarts in background again
4 it takes some seconds (2-4) until the captured image is displayed over the live preview

### Expected behavior

Show immediately

### Photobooth Version

from today

### Browser and browser version

chromium latest bullseye version

### Operating System

RPI os Bullseye

### Did it work before?

Yes

### Latest working version

not very sure.

andi34, 2022-11-06:
I guess you're talking about an really old version. In the past the image filter had to be defined **before** an Image was taken. We've changed that for more flexibility, the filter now get applied after the image was taken.

You should make sure __Preload and show image during filter processing__ is disabled 

Sure , we can discuss to use the thumbnail on result screen instead showing the full sized image. This should be easy to implement as an option.

I'll take a closer look next days, maybe we can improve more.

mgrl, 2022-11-06:
`You should make sure Preload and show image during filter processing is disabled`
Was disabled already.

Maybe you can point me to some lines I may test to comment out or fiddle with and check whether it speeds up?

andi34, 2022-11-06:
No , not directly. I'll taking a closer look at the `api.processPic` and `api.renderPic` functions when I find some time.

andi34, 2022-11-07:
I guess you see the image Multiple times because it's once processed via `/api/takePic.php` (image taken) and once again via `/api/applyEffects.php` (image post processing incl creating the thumbnail).
After all that, the image gets added to the gallery via the core.js, also loaded to be visible on the result screen.

mgrl, 2022-11-07:
Is applyEffects processing the image in any way (despite creating a thumbnail) even if I do not want any filter to be applied?  The URL ends with xxx.jpg?filter=plain - so the image is processed in any way we can avoid?

closed 2023-01-14 by andi34