PhotoboothProject/photobooth #99: [BUG]: yarn install Segmentation Fault

CaptainCookLP, 2022-12-15:
### Describe the bug

### Now we are going to install Photobooth.
Cloning into 'html'...
remote: Enumerating objects: 14784, done.
remote: Counting objects: 100% (1213/1213), done.
remote: Compressing objects: 100% (506/506), done.
remote: Total 14784 (delta 822), reused 1013 (delta 692), pack-reused 13571
Receiving objects: 100% (14784/14784), 11.09 MiB | 4.25 MiB/s, done.
Resolving deltas: 100% (10132/10132), done.
### Checking needed remote information...
### Adding photoboothproject remote...
### We are installing/updating Photobooth via git.
From https://github.com/PhotoboothProject/photobooth
 * branch            dev        -> FETCH_HEAD
 * [new branch]      dev        -> photoboothproject/dev
Note: switching to 'photoboothproject/dev'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at 40f10a6 bump version to v4.3.1
Submodule 'vendor/PHPMailer' (https://github.com/PHPMailer/PHPMailer) registered                                                                                                                                                                                                                                              for path 'vendor/PHPMailer'
Submodule 'vendor/PhotoSwipe' (https://github.com/PhotoboothProject/PhotoSwipe)                                                                                                                                                                                                                                              registered for path 'vendor/PhotoSwipe'
Submodule 'vendor/Seriously' (https://github.com/brianchirls/Seriously.js) regis                                                                                                                                                                                                                                             tered for path 'vendor/Seriously'
Submodule 'vendor/phpqrcode' (https://github.com/PhotoboothProject/phpqrcode) re                                                                                                                                                                                                                                             gistered for path 'vendor/phpqrcode'
Cloning into '/var/www/html/vendor/PHPMailer'...
Cloning into '/var/www/html/vendor/PhotoSwipe'...
Cloning into '/var/www/html/vendor/Seriously'...
Cloning into '/var/www/html/vendor/phpqrcode'...
Submodule path 'vendor/PHPMailer': checked out '8b6386d7417526d1ea4da9edb70b8352                                                                                                                                                                                                                                             f7543627'
Submodule path 'vendor/PhotoSwipe': checked out 'ac3253927eab72fd8b62310d88012fb                                                                                                                                                                                                                                             f9927fc7e'
Submodule path 'vendor/Seriously': checked out '513d2b8696b76fe764b3033aa6dc7c99                                                                                                                                                                                                                                             005acccd'
Submodule path 'vendor/phpqrcode': checked out '863ffffac4c9d22e522464e325cbcbad                                                                                                                                                                                                                                             fbb26470'
### Get yourself a hot beverage. The following step can take up to 15 minutes.
yarn install v1.22.19
[1/5] Validating package.json...
[2/5] Resolving packages...
[3/5] Fetching packages...
[-----------------------------------------------------------------------] 0/674i                                                                                                                                                                                                                                             nstall-photobooth.sh: line 561: 15362 Segmentation fault      yarn install

### Steps to reproduce the behavior

Installation with the Auto Installation Script on Rasberry Pi OS bullseye

Updated till 15.12.22 15:30 Berlin/Europe Time

### Expected behavior

Working Fine

### Photobooth Version

4.3.1

### Did it work before?

Yes

andi34, 2022-12-16:
Hey and thanks for your Report!

I am not sure why you're having issues, maybe there's a hardware issue (limited RAM?)? Could you please tell more about your hardware setup?

To avoid the issue, you could install Photobooth from latest stable package running `sudo bash install-photobooth.sh -username='<YourUsername>' -branch='package'`.

CaptainCookLP, 2022-12-16:
The Hardware is an old Acer Notebook with an i5 and 4GBs of RAM.
Installed ist the latest Raspberry Pi OS for i386

I will try the Installation with the latest stable

CaptainCookLP, 2022-12-16:
Installation Process now was succesful.

Next Problem, my Camera (Canon EOS 550D) has no output in the LivePreview, Taking Pictures is Working fine.
The GPhoto2 Documentation about supported Cameras says, it has the Feature.

Preview is enabled (from Device CAM)

Webcam of Notebook is disabled (/etc/modprobe.d/blacklist.conf -> blacklist uvcvideo)

andi34, 2022-12-16:
Please take a look at https://github.com/PhotoboothProject/photobooth/issues/80

CaptainCookLP, 2022-12-16:
python3 cameracontrol.py --bsm is working fine, but still no automatic preview.

If I start this command manually in the terminal it's working fine. With no issues.

I know I'm a little annoying, I'm sry for that. My current config is down below

`Array
(
    [ui] => Array
        (
            [language] => de
            [folders_lang] => /resources/lang
            [show_fork] => 
            [skip_welcome] => 1
            [style] => modern
            [button] => modern
            [shutter_animation] => 1
            [shutter_cheese_img] => /private/cheese.png
            [result_buttons] => 1
            [font_size] => 16px
            [decore_lines] => 1
            [rounded_corners] => 
            [github] => PhotoboothProject
            [branding] => Photobooth
        )

    [adminpanel] => Array
        (
            [view] => expert
            [view_default] => expert
        )

    [dev] => Array
        (
            [loglevel] => 2
            [demo_images] => 
            [reload_on_error] => 1
            [logfile] => error.log
        )

    [start_screen] => Array
        (
            [title] => FKmedify
            [title_visible] => 1
            [subtitle] => Photobooth
            [subtitle_visible] => 1
        )

    [picture] => Array
        (
            [thumb_size] => 540px
            [time_to_live] => 20
            [preview_before_processing] => 
            [retry_on_error] => 0
            [retry_timeout] => 2
            [cntdwn_time] => 3
            [cntdwn_offset] => 0
            [no_cheese] => 
            [cheese_time] => 500
            [flip] => off
            [rotation] => 0
            [polaroid_effect] => 
            [polaroid_rotation] => 0
            [take_frame] => 1
            [frame] => /var/www/html/resources/img/frames/frame.png
            [key] => 
            [naming] => dateformatted
            [permissions] => 0644
            [keep_original] => 1
            [preserve_exif_data] => 
[... 462 lines cut ...]
        (
            [images] => data/images
            [keying] => data/keying
            [print] => data/print
            [qrcodes] => data/qrcodes
            [thumbs] => data/thumbs
            [tmp] => data/tmp
        )

    [foldersJS] => Array
        (
            [images] => /data/images
            [keying] => /data/keying
            [print] => /data/print
            [qrcodes] => /data/qrcodes
            [thumbs] => /data/thumbs
            [tmp] => /data/tmp
        )

    [photobooth] => Array
        (
            [version] => 4.3.1
        )

)`

CaptainCookLP, 2022-12-16:
My bad, evverything is fine. I just had to read the Documentation a bit more harder

closed 2022-12-16 by CaptainCookLP