PhotoboothProject/photobooth #108: [FEATURE]: Upload to Cloud

CaptainCookLP, 2022-12-19:
### Is your feature request related to a problem?

No

### Description

Implementation of this Plugin to Upload directly to Dropbox
https://github.com/andreafabrizi/Dropbox-Uploader

### Describe the solution you'd like

Easy Implementation via Admin Panel of the Upload Function without Configuration via CLI

### Describe alternatives you've considered

None

andi34, 2022-12-19:
Hey and thanks for your suggestion!
Currently I have only limited time to implement new features, but sure I'll find time to review a Pullrequest if someone likes to implement it.

@fmiccolis already started working on something
https://github.com/fmiccolis/photobooth/tree/feature/ftp

Best regards

andi34, 2022-12-19:
Btw, since the filename is available via post-command, you can easily upload via CMD using the post command option https://github.com/PhotoboothProject/photobooth/commit/14eea7c18c438ac55b87baff5758a3f7d73063af

remylpat, 2023-01-30:
I think Push to cloud service is an important option to consider.

I was digging a way to upload on a GooglePhoto Album that could be share to all during a Photobooth Event.

I try re-used the script from : https://github.com/eshmu/gphotos-upload and the Post-CMD 
I can upload from cli using gphotos-upload but can not from the Post-Cmd.

It is as simple as a python file to store on the good folder and a Post-Cmd to define but I'm still failing as I am not a developer anymore.

andi34, 2023-02-01:
Sounds like a permission error you get and you should try to fix. Make sure pipenv is installed for all user: `pip install pipenv` (for reference see https://stackoverflow.com/questions/36898474/how-to-install-a-module-for-all-users-with-pip-on-linux )

You can use a bash script you run from post command which can be more complex.

As post command e.g. use `bash /var/www/html/private/upload.sh`

The upload.sh could look like this:

```sh
#!/bin/bash
#
PICTURE="$1"

if [ ! -d "/var/www/html/private/gphotos-upload" ]; then
  git clone https://github.com/eshmu/gphotos-upload.git
fi

cd /var/www/html/private/gphotos-upload
pipenv install
pipenv she'll
python3 upload.py
exit 0
```
(Not tested)

andi34, 2023-02-01:
It definitely work with scp to upload to a remote server, ssh keys need to be created as www-data user.

Works like that:
```sh
scp [OPTION] [user@]SRC_HOST:]file1 [user@]DEST_HOST:]file2
```
https://linuxize.com/post/how-to-use-scp-command-to-securely-transfer-files/

You could also call a bash script from post-command for that and pass the filename in there to reuse (see previous comment)
```sh
bash /var/www/html/private/upload.sh %s
```

As you see there's ways without the need of a direct implemention inside Photobooth.

gphotos-upload would never be an option to include because of the missing license information.

remylpat, 2023-02-01:
thx andi34, I'm agree it is sad there is no licences info on that gphotos-upload script

closed 2023-02-02 by andi34

andi34, 2023-02-02:
Not sure if this still works

https://olivermarshall.net/how-to-upload-a-file-to-google-drive-from-the-command-line/

If someone is trying: don't forget to setup as www-data user (running the commands with `sudo -u www-data`).