PhotoboothProject/photobooth #117: [BUG]: Cheese image is no longer displayed

obkram, 2023-01-05:
### Describe the bug

#german
Seit dem Update von Photobooth Version 3.3.0 auf 4.3.1 wird zwischen dem Countdown und dem Auslösen der Kamera das Cheese oder in meinem Fall das Bild mit der Kamera und dem Pfeil nach oben nicht mehr angezeigt. Ich habe danach auch im Ordner /privat die Datei cheese.png eingefügt und über das Admin Panel ausgewählt. Der Bildschirm bleibt in diesem Zeitraum leider weiter schwarz. Hat dieses Problem noch irgendwer außer mir?

#english
Since the update of Photobooth version 3.3.0 to 4.3.1, the cheese or in my case the image with the camera and the up arrow no longer appears between the countdown and the triggering of the camera. I then also inserted the file cheese.png in the /private folder and selected it via the admin panel. Unfortunately, the screen remains black during this period. Does anyone else have this problem besides me?

### Steps to reproduce the behavior

#german
Cheeseeinblendung zwischen Countdown and Auslösen des Bildes

#english
Cheese overlay between countdown and triggering the image

### Expected behavior

#german
1. Countdown
2. Cheese Bild
3. Aufnahme des Bildes

#english
1. Countdown
2. Cheese picture
3. Picture taking picture

### Photobooth Version

4.3.1

### Browser and browser version

IPad Safari, App Frameless, Safari Iphone 11

### Operating System

Raspbian 11 bullseye

### Did it work before?

Yes

### Latest working version

3.3.0

andi34, 2023-01-05:
Thanks for the report! Could you please test https://github.com/PhotoboothProject/photobooth/pull/118 ?

obkram, 2023-01-05:
Hi Andi, I have customised the scr/core.js accordingly in line 385 with the effect that the cheese sequence not only remains black but is completely missing.

andi34, 2023-01-05:
Thanks. Let me check later today or tomorrow. I'll let you know once there's something new to test

obkram, 2023-01-05:
Thank you for troubleshooting. Possibly this really only affects my system and is not a general problem.

obkram, 2023-01-05:
And this is what it looks like…

https://user-images.githubusercontent.com/33160929/210812400-f5bb6441-b441-410f-84b0-6eac1f59baca.MOV

andi34, 2023-01-05:
For me it works with mentioned change. Make sure to run a "yarn build" after applying the change. Also maybe increase the cheese time slightly.

closed 2023-01-06 by andi34