PhotoboothProject/photobooth #124: [BUG]: Installation on Windows don't work fine

HappySeppel, 2023-01-13:
### Describe the bug

After following the installation instructions for windows, the admin panel and all other sites dont look how expected.
Without background and structure.
screenshot: [https://1drv.ms/u/s!AiwFcU1xxp78mhjYUo13iV0CjOwK?e=oo74ey](url)

How can I solve this problem?

### Steps to reproduce the behavior

follow the install instructions.

### Expected behavior

appear like the normal frontend

### Photobooth Version

4.3.1

### Browser and browser version

Chrome and Firefox

### Operating System

Win 10

### Did it work before?

Yes

andi34, 2023-01-13:
Please open the browser console (F12) and check for error entries. Screenshots can be uploaded to the comment here too (preferred).

You've always used the port and subfolder path in the past?

andi34, 2023-01-13:
Also it really helps, if you remember which version was used before and worked. This might make it easier to find out what could make a difference.

andi34, 2023-01-13:
Could you please edit the `lib/helper.php` and make the following changes?

Change

```php
function isSubfolderInstall() {
    $path = getrootpath(__DIR__ . DIRECTORY_SEPARATOR . '..' . DIRECTORY_SEPARATOR);
    $server_path = getPhotoboothPath();

    if ($path == $server_path) {
        return false;
    } else {
        return true;
    }
}
```

To

```php
function isSubfolderInstall() {
    return true;
}
```
Save the file.

Now please rename the `config/my.config.inc.php` to `config/my.config.inc.php.bak`.

Test again.

andi34, 2023-01-13:
Another Simple question:.you've used the `photobooth-4.3.1.zip` or `Source code `(zip)?  Only the `photobooth-4.3.1.zip` contains everything needed to work on windows.

HappySeppel, 2023-01-14:
Thank you for your support. I downloaded now photobooth-4.3.1.zip and now the main structure is showing but not all components and styling details. By checking the console for errors I found this. The folder booth will be called between the resources.

`[ 2023-01-14T08:52:56.900Z ]: Loglevel: 1
/booth/admin/booth/resources/lang/en.json:1          Failed to load resource: the server responded with a status of 404 (Not Found)
utils.js:50 Could not fetch "http://localhost/booth/admin/booth/resources/lang/en.json": 404 (Not Found)

This error happened in the method `    at r.debug (http://localhost/booth/node_modules/` from: `andreasremdt/simple-translator/dist/umd/translator.min.js:1:3761)`.

If you don't want to see these error messages, turn off debugging by passing `{ debug: false }` to the constructor.

Error code: FETCH_ERROR

Check out the documentation for more details about the API:
https://github.com/andreasremdt/simple-translator#usage
        
(anonymous) @ utils.js:50
utils.js:50 No translation for language "en" has been added, yet. Make sure to register that language using the `.add()` method first.

This error happened in the method `    at r.debug (http://localhost/booth/node_modules/` from: `andreasremdt/simple-translator/dist/umd/translator.min.js:1:3761)`.

If you don't want to see these error messages, turn off debugging by passing `{ debug: false }` to the constructor.

Error code: NO_LANGUAGE_REGISTERED

Check out the documentation for more details about the API:
https://github.com/andreasremdt/simple-translator#usage
        
(anonymous) @ utils.js:50
bg_stone.jpg:1          Failed to load resource: the server responded with a status of 404 (Not Found)`

andi34, 2023-01-14:
Adjust the language resources path config to your needs.

HappySeppel, 2023-01-14:
I change the document root in apache on the main folder and now it works fine. Thanks for your help.

andi34, 2023-01-14:
Thanks for the feedback!
I'll take a closer look later this weekend. Seems to be only an issue on subfolder install which I didn't faced on Linux.

Can I send you a zip once I got something to test again on a subfolder install?

HappySeppel, 2023-01-14:
Sure, I will test it and give you feedback.

andi34, 2023-01-14:
https://drive.google.com/file/d/1T4mmAbgnavdBbtzUTFhMLZTgv0zUK0Qf/view?usp=drivesdk

Make sure there's no my.config.inc.php inside the config folder while testing again. There's a few upcoming restructure changes applied on that version. Would be great if you can test on a subfolder Installation.

HappySeppel, 2023-01-14:
Hi, I tested the installation but it doesn't work. Attached you see the Errors from the console:

`translator.js:299          GET http://fotobox-hp:81/C:/Apache24/htdocs/photobooth/resources/lang/en.json 403 (Forbidden)
(anonymous) @ translator.js:299
(anonymous) @ translator.js:299
e @ translator.js:267
(anonymous) @ i18n.js?v=4.3.1:12
utils.js:50 Could not fetch "http://fotobox-hp:81/C:/Apache24/htdocs/photobooth/resources/lang/en.json": 403 (Forbidden)

This error happened in the method `    at r.debug (http://fotobox-hp:81/node_modules/` from: `andreasremdt/simple-translator/dist/umd/translator.min.js:1:3761)`.

If you don't want to see these error messages, turn off debugging by passing `{ debug: false }` to the constructor.

Error code: FETCH_ERROR

Check out the documentation for more details about the API:
https://github.com/andreasremdt/simple-translator#usage
        
(anonymous) @ utils.js:50
(anonymous) @ translator.js:307
(anonymous) @ translator.js:302
Promise.then (async)
(anonymous) @ translator.js:300
e @ translator.js:267
(anonymous) @ i18n.js?v=4.3.1:12
utils.js:50 No translation for language "en" has been added, yet. Make sure to register that language using the `.add()` method first.

This error happened in the method `    at r.debug (http://fotobox-hp:81/node_modules/` from: `andreasremdt/simple-translator/dist/umd/translator.min.js:1:3761)`.

If you don't want to see these error messages, turn off debugging by passing `{ debug: false }` to the constructor.

Error code: NO_LANGUAGE_REGISTERED

Check out the documentation for more details about the API:
https://github.com/andreasremdt/simple-translator#usage
        
(anonymous) @ utils.js:50
value @ translator.js:160
(anonymous) @ i18n.js?v=4.3.1:13
Promise.then (async)
(anonymous) @ i18n.js?v=4.3.1:12
modern_squared_admin.css:1          GET http://fotobox-hp:81/C:/Apache24/htdocs/photobooth/resources/img/bg_stone.jpg 403 (Forbidden)
tools.js?v=4.3.1:12 [ 2023-01-14T14:58:29.160Z ]: Loglevel: 1`

andi34, 2023-01-14:
Thank!
Again please delete the my.config.inc.php

Replace the content of the lib/config.php with the following:

https://raw.githubusercontent.com/PhotoboothProject/photobooth/restructure/classes/lib/config.php

Does it work with that?

HappySeppel, 2023-01-15:
Today I'm going to business journey, I test it when I'm back.

andi34, 2023-01-16:
Hey!
I was able to test myself on windows and there's still an issue. I'll look into it myself when I get some more time. I'll let you know if I need someone else to cross-check it's working once I am done with it 

Thanks a lot!

closed 2023-04-15 by andi34