PhotoboothProject/photobooth #135: [BUG]: translated button does not fit

mart-e, 2023-02-13:
### Describe the bug

When using the photobooth in French, there is a few terms that do not fit the buttons box
[image]
[image]

Is it something to fix at the translation levels (I am ready to help) or at the CSS (not sure here) ?

### Steps to reproduce the behavior

1. Switch the interface to French
2. Go to home screen, take a picture,...

### Expected behavior

Everything fits the UI

### Photobooth Version

4295699a739cc50eee2aae7fedc54a21d36ee714

### Browser and browser version

Firefox 109

### Operating System

Raspbianlite

### Did it work before?

No

andi34, 2023-02-13:
Ia it possible to use a shorter text? If yes please join us at crowdin and suggest a new translation.
https://crowdin.com/project/photobooth

mart-e, 2023-02-13:
I am already member of the crowdin team, will suggest translations :)
I just wanted to be sure if we try to fix it at the translation level or css.

When are the translation fetched ? I see the "Collage -> Photomontage" string was changed one month ago (probably for that reason).
https://crowdin.com/translate/photobooth/2/en-fr#1286

Side-question : is it best I run on the dev version (as offered by the [install instructions](https://photoboothproject.github.io/Installation-on-Debian)) or the latest stable ?

andi34, 2023-02-13:
I am trying to sync the translations next days. Currently the monthly import via GitHub is broken and need to be fixed... Need some time to investigate.

Latest dev branch always contains improvements, sometimes new features and small Bugfixes for latest stable version.
Currently there's some changes I'd like to finish to place a new stable release, but private life had some other plans for me last week's... Hope to get things sorted this month.

andi34, 2023-02-13:
But if you like to adjust the css too, I'd be glad to review a Pullrequest with needed changes.

mart-e, 2023-02-13:
No problem, take your time, just wanted to be sure I contributed at the right place.
Thanks for making that project :)

closed 2023-02-13 by mart-e