PhotoboothProject/photobooth #144: [BUG]: Live View - Nikon D5200

koerby, 2023-03-16:
### Describe the bug

cameracontrol.py - stops with error

```
[mjpeg @ 0x7fcd50] Format mjpeg detected only with low score of 25, misdetection possible!
Input #0, mjpeg, from 'pipe:':
  Duration: N/A, bitrate: N/A
    Stream #0:0: Video: mjpeg (Baseline), yuvj422p(pc, bt470bg/unknown/unknown), 640x424, 25 tbr, 1200k tbn, 25 tbc
Codec AVOption preset (Configuration preset) specified for output file #0 (/dev/video0) has not been used for any stream. The most likely reason is either wrong type (e.g. a video option with no video streams) or that it is a private option of some encoder which was not actually used for any stream.
Stream mapping:
  Stream #0:0 -> #0:0 (mjpeg (native) -> rawvideo (native))
[swscaler @ 0x81d200] deprecated pixel format used, make sure you did set range correctly
[video4linux2,v4l2 @ 0x805780] ioctl(VIDIOC_G_FMT): Invalid argument
Could not write header for output file #0 (incorrect codec parameters ?): Invalid argument
Error initializing output stream 0:0 -- 
Conversion failed!
Traceback (most recent call last):
  File "/var/www/html/api/cameracontrol.py", line 295, in <module>
    main()
  File "/var/www/html/api/cameracontrol.py", line 291, in main
    CameraControl(args)
  File "/var/www/html/api/cameracontrol.py", line 35, in __init__
    self.pipe_video_to_ffmpeg_and_wait_for_commands()
  File "/var/www/html/api/cameracontrol.py", line 212, in pipe_video_to_ffmpeg_and_wait_for_commands
    self.ffmpeg.stdin.write(img_bytes)
BrokenPipeError: [Errno 32] Broken pipe
```

v4l2loopback - 0.12.7
photobooth - 4.3.1

### Steps to reproduce the behavior

I try this: gphoto2 --stdout --capture-movie | ffmpeg -i - -vcodec rawvideo -pix_fmt yuv420p -threads 0 -f v4l2 /dev/video0

but the same problem:

[mjpeg @ 0x1e4ed50] Format mjpeg detected only with low score of 25, misdetection possible!
Input #0, mjpeg, from 'pipe:':
  Duration: N/A, bitrate: N/A
    Stream #0:0: Video: mjpeg (Baseline), yuvj422p(pc, bt470bg/unknown/unknown), 640x424, 25 tbr, 1200k tbn, 25 tbc
Stream mapping:
  Stream #0:0 -> #0:0 (mjpeg (native) -> rawvideo (native))
[swscaler @ 0x1e71a90] deprecated pixel format used, make sure you did set range correctly
[video4linux2,v4l2 @ 0x1e57780] ioctl(VIDIOC_G_FMT): Invalid argument
Could not write header for output file #0 (incorrect codec parameters ?): Invalid argument
Error initializing output stream 0:0 -- 
Conversion failed!

### Expected behavior

+

### Photobooth Version

4.3.1

### Browser and browser version

Chromium Version 109.0.5414.112

### Operating System

Raspberry Pi 4 Model B Rev 1.1 | Raspbian GNU/Linux 11 (bullseye) with desktop

### Did it work before?

No

andi34, 2023-03-17:
Camera should be set to one of these: P/S/A/M
What's set on your camera?

koerby, 2023-03-26:
Hi, sorry for the late answer.
Camera was at M, think the image musst be set to jpg not raw, now it works.

andi34, 2023-03-26:
Thanks for the feedback!

closed 2023-03-26 by andi34