PhotoboothProject/photobooth #147: [FEATURE]: Allow continueing preview from device cam even if the take photo command gets send to camera

s5919, 2023-03-26:
### Is your feature request related to a problem?

Yes

### Description

When using the preview via device cam (e.g. tablet) and at the same time setting the "Countdown Offset" for example to 3 seconds. The preview will be stopped 3 seconds before the end of the countdown. This is only necessary if the preview and the image is provided by the camera. In case of using a device cam it would be possible to continue the preview until the countdown is done.

### Describe the solution you'd like

Avoid stopping the preview if "device cam" is used until the countdown has finished.

### Describe alternatives you've considered

I am not sure if that would be possible but if the "stop preview offset" could be set to a negative value it could may fix the issue (except if the "take picture command" is always stopping the preview).

s5919, 2023-03-27:
With the support from @andi34 I was able to find a workaround for Windows until this is fixed:

1. Open a command window and change the output folder of the digicamcontrol
`C:\Apache24\htdocs\digicamcontrol\CameraControlRemoteCmd.exe /c set session.folder C:\Apache24\htdocs\data\tmp`
2. Change the "Take picture command" in the admin area to the following
`C:\Apache24\htdocs\digicamcontrol\CameraControlRemoteCmd.exe /c capture %s`
3. Update the takePic.php file as shown [here](https://github.com/PhotoboothProject/photobooth/pull/148/files)
[image]

andi34, 2023-03-27:
1. Define `C:\Apache24\htdocs\digicamcontrol\CameraControlRemoteCmd.exe /c set session.folder C:\Apache24\htdocs\data\tmp` as pre-Photo command
2. `C:\Apache24\htdocs\digicamcontrol\CameraControlRemoteCmd.exe /c capture %s` is the suggested usage by the digiCamControl maintainer
3. Merged into dev branch already. Also adjusted to only log on loglevel > 1