PhotoboothProject/photobooth #150: [BUG]: Live Preview - Camera Module 3

mikemyers777, 2023-03-28:
### Describe the bug

With Raspberry Camera Module 3, the live preview is not working.
Also with motion and the settings described in the Faq's.

Is there OS adjustments required?

### Steps to reproduce the behavior

Use Camera Module 3 

### Expected behavior

Working Live preview 

### Photobooth Version

4.3.1

### Browser and browser version

Chromium/latest Raspberry OS

### Operating System

Raspberry OS

### Did it work before?

No

andi34, 2023-03-29:
Hey and thanks for the report. It's an issue/limitation of the Rapsberry pi Kernel and not an Photobooth issue. Therefore closing the issue here.

To solve your problem you should install https://github.com/PhotoboothProject/photobooth-app too and follow the Photobooth config mentioned inside the README.

Best regards

Andi

closed 2023-03-29 by andi34

mikemyers777, 2023-03-30:
Hi Andi! Thx for the fast help and info. 
I tried to solve the problem with the description from the link, but running in problems during the installation..
" run cmd: 'pip install -r requirements.txt', sudo=False
╚══════════════════════════
ERROR: Could not open requirements file: [Errno 2] Datei oder Verzeichnis nicht gefunden: 'requirements.txt'
command exited with returncode= 1
pip installation failed! check output for errors
"
Could you help me eith this?

Br,
Michael

mikemyers777, 2023-04-11:
Great, thank you very much! It's working now 👍