PhotoboothProject/photobooth #202: [BUG Pi OS]: v4l2loopback-dkms fails to install on Pi OS

andi34, 2023-05-05:
### Describe the bug

Installation fails because v4l2loopback module can't be compiled.

### Steps to reproduce the behavior

Start installation via Installer.

### Expected behavior

no trouble 

### Photobooth Version

Independent 

### Operating System

PiOS

### Did it work before?

Yes

### Latest working version

Unrelated to Photobooth 

### Additional context

Need to be fixed via official Pi Sources. See https://github.com/raspberrypi/linux/issues/5410

Workaround on Pi OS 32bit:
```
sudo nano /boot/config.txt 
```
Add
``` 
arm64=0
```

And reboot once before starting the installation.

semikolijon, 2023-05-15:
"arm64=0" don't work for me - instead "arm_64bit=0" works fine.

mart-e, 2023-05-31:
> Additional context
> 64bit PiOS should be used.

Note that I had other issues with 64bits, for instance the pyhon picamera module does not seem to be compatible with 64 bits (cf https://github.com/raspberrypi/firmware/issues/1558)

The workaround did work for me.

andi34, 2023-05-31:
Thanks for the information, we need to wait for the Pi foundation to get things fixed.

weimer92, 2023-06-27:
> "arm64=0" don't work for me - instead "arm_64bit=0" works fine.

The workaround did work for me.

mart-e, 2023-08-12:
Hi again,

> Note that I had other issues with 64bit

Following a discussion on photobooth-app (mgrl/photobooth-app#76), I gave it another try using amd64... and did not get any issue. I installed both PhotoboothProject and photobooth-app and so far, everything seems to work fine.

So yes, maybe going 64bits instead of 32 is a good recommendation ?

reloxx13, 2026-11-06:
This report concerns the Raspberry Pi kernel/header/DKMS combination rather than Photobooth application code. The linked [upstream discussion](https://github.com/raspberrypi/linux/issues/5410) identifies missing matching kernel headers and explains the 32-bit userspace / 64-bit kernel case.

An important limitation of the historical workaround: `arm_64bit=0` applies only to a **32-bit OS installation**. It must not be applied to a 64-bit OS installation; the upstream discussion includes a boot failure and explains why a 32-bit kernel cannot run 64-bit applications.

For anyone still affected on a current system, please provide the Raspberry Pi OS release, `uname -a`, `getconf LONG_BIT`, the installed kernel/header package versions, and the relevant DKMS `make.log`. I have not reproduced the old kernel issue on current Pi OS, so this is not a claim that it is fixed today. There is no verified Photobooth source change to propose from the existing evidence.