PhotoboothProject/photobooth #227: [FEATURE]: Print Support

capsload2, 2023-06-13:
### Is your feature request related to a problem?

No

### Description

Is there a way to print the snapshits directly with a supported printer?

### Describe the solution you'd like

Support printers

### Describe alternatives you've considered

—-

andi34, 2023-06-13:
Yes. On linux you can use CUPS to setup your printer. Photobooth can execute the print command via `lp` command. Example command is added by default.

It's an option to print the image immediately after it was taken, else you can enable a print Button as needed.

How to setup a printer isn't part of Photobooth. If you need any help I am sure someone is able to help you in our telegram group.

andi34, 2023-06-13:
[image]

closed 2023-06-13 by andi34