PhotoboothProject/photobooth #234: [FEATURE]: Bulk download for photos in gallery

floviolleau, 2023-06-22:
### Is your feature request related to a problem?

No

### Description

Hi !

Thanks for this amazing project!
I was able to setup it in less than 3 days.
I made a small script to disable all keyboard keys (except space) to leave it unattended.

Some users asked me to be able to bulk download photos in gallery

Thanks a lot!

### Describe the solution you'd like

Bulk download of photos in gallery
Can be checkboxes and a download button that zip all selected photos.

### Describe alternatives you've considered

I used a Nextcloud plugged to the folder of the photobooth project and a cron to update the db.txt file
So that people can bulk download and can drop photos of the event from their phone at the same time.

I used a small router to make an access point with pihole installed on the machine where photobooth was installed to have a local dns for the gallery and the nextcloud
People download either from a "beautiful" name or a QRCode

andi34, 2023-07-01:
https://github.com/PhotoboothProject/photobooth/pull/208

Nextcloud integration is WIP and should be available on next release.

Rudi-Ment, 2023-07-17:
Hi.
At first: Thank you for your very nice project. 
I made a photobooth for the weeding in our family and for the birthdays of our kids.
The responce is allways great !

Is it possible -when you implement the Nextcloud adaptation- to install also an access to PCloud ?
I think PCloud is also a good Cloud-Service...

Best regards

andi34, 2023-07-17:
Hey and thanks for your request. I am not going to add more cloud services myself. If someone of the community likes to add it I am glad to review needed changes for it. The Nextcloud integration also is a community contribution.

Best regards

Andi

floviolleau, 2023-09-18:
Hi,

My query is not to add Nextcloud into Photobooth project. It was the alternative I used.
I ask only to allow bulk download in gallery. Please re-read the issue description

Have a good day

fmalcher, 2024-01-29:
This issue went off as a request for Nextcloud integration. However, coming back to the original idea:
**Could we support a bulk download?** At the moment the only two ways of getting all pictures is through USB Sync or via direct file system access with SSH. Is there anything else I don't know? Thanks!

andi34, 2024-01-29:
There was an quite experimental option to download the whole data folder via disk usage page (access via Adminpanel), but I think this got lost while switching to tailwind-css for the admin UI.

For now you could create an own page inside private/ folder of Photobooth to use the endpoint

https://github.com/PhotoboothProject/photobooth/blob/dev/api/diskusage.php

andi34, 2024-01-29:
Reopened, maybe something to add in future.

floviolleau, 2024-06-03:
Unstale

closed 2024-08-16 by stale[bot]