PhotoboothProject/photobooth #235: [BUG]: Pi Camera not working after completing the installation of the Photobooth system

colaplate, 2023-06-24:
### Describe the bug

My Raspberry Pi basic environment is 3B+, and I installed Bullseye -ARM64-lite. Before running the install-photobooth script, I made some necessary configurations for the pi-camera (V1.3 ov5647), and the pi-camera was working fine. However, after running the install-photobooth script, the web service of the Photobooth started successfully, but the pi-camera stopped working.

Please note that this is a translation and may not capture the exact meaning or tone of the original text. It is always recommended to provide the original text along with the translation when communicating with developers for better understanding and accuracy.

### Steps to reproduce the behavior

I performed standard updates and executed "sudo rpi-update".
Run raspi-config, I enabled "legacy camera support", disabled the "GL driver", and enabled "glmor".
sudo nano /boot/config.txt, I confirmed the following settings: dtparam=i2c_arm=on, dtparam=i2s=on, start_x=1, #dtoverlay=vc4-kms-v3d, dtoverlay=vc4-fkms-v3d, and dtoverlay=ov5647.
After completing the above steps,  reboot.

run "vcgencmd get_camera" shows "supported=1 detected=0, libcamera interfaces=1".
run "libcamera-still -n -o %s -q 100 -t 1 | echo Done" provides camera information feedback.
run"libcamera-still -o test.jpg" works fine.

after running the install-photobooth script,the web service of the Photobooth is running properly and can be accessed, 
but at this point, the pi-camera stops working.
run "libcamera-still -n -o %s -q 100 -t 1 | echo Done" 
feedback:
Done
[0:04:47.945102977] [794] INFO Camera camera_manager.cpp:299 libcamera v0.0.4+22-923f5d70
ERROR: *** no cameras available ***

### Expected behavior

How to properly configure pi-camera support

### Photobooth Version

The query shows that it is the latest version, installed through the installation script.

### Did it work before?

No

colaplate, 2023-06-24:
Furthermore, the FAQ mentions the following information:

"Once done, you need to adjust the configuration. Open the admin panel in your browser at localhost/admin and make the following changes:
'Take picture command on Pi OS based on bullseye':
libcamera-still -n -o %s -q 100 -t 1 | echo Done"

However, I cannot find any configurable content in localhost/admin. Is there a more specific section or location for this configuration?

andi34, 2023-06-24:
I get you installed the service for preview from DSLR , those disables the pi camera.

Run

```
sudo bash /var/www/html/gphoto/install-gphoto-python.sh`

closed 2023-06-24 by andi34

mart-e, 2023-07-14:
Could it be the same issue as this one?
raspberrypi/libcamera-apps#534
To fix it, I had to add `camera_auto_detect=1` in my `/boot/config.txt` file