PhotoboothProject/photobooth #256: [BUG]: Undefined Variable in keypad screen

Gefaehrbert, 2023-07-15:
### Describe the bug

When i open the Keypad to log in i get this error:
[image]

### Steps to reproduce the behavior

Open the keypad

### Expected behavior

no error 

### Photobooth Version

4.3.1 

### Operating System

ubuntu

### Did it work before?

Yes

andi34, 2023-07-15:
You're on the WIP branch?

andi34, 2023-07-15:
https://github.com/PhotoboothProject/photobooth/commit/c888a4818cc4b6719911f622d3dfc3a896995d36

This should fix it

PR #257 (2023-07-15): login(keypad): fix undefined $fileRoot variable
#### Prerequisites checklist

- [x] I have read the [contributing guidelines](https://github.com/PhotoboothProject/photobooth/blob/dev/CONTRIBUTING.md).

#### What is the purpose of this pull request? (put an "x" next to an item)

- [ ] Documentation update
- [x] Bug fix
- [ ] New feature
- [ ] Other, please explain:

#### What changes did you make? (Give an overview)
Fix #256 

#### Is there anything you'd like reviewers to focus on?
closed 2023-07-15 by andi34