PhotoboothProject/photobooth #264: [FEATURE]: allow HID remote button on photobooth server

mutefiRe, 2023-07-30:
### Is your feature request related to a problem?

Yes

### Description

I use an Android Tablet as a Screen connected with bluetooth button which hits the key code `Volume UP`. The Volume up has to be remapped laboriouslly to another key because the browser does not recognise it otherwise. This does not work well and for the remapping you have to swap from original keyboard layout to the keymapping app layout, which totally elimates typing for photo booth users, e.g. typing in their email address. 

 

### Describe the solution you'd like

I would prefer connecting the bluetooth device with e.g. the raspberry Pi where the Photobooth server is running and listen there for the the key code to trigger whatever is configured. This would be a more centralized solution to eliminate external device configuration. 

### Describe alternatives you've considered

As it's too error prone I would just use the UI buttons on the tablet screen.

### Additional context

If you give this a go I would like to contribute on this nice project :)

andi34, 2023-07-30:
Hey!

I am just about to seperate the remoteBuzzer server optional and allow to use the remoteBuzzer server standalone, e.g. on a Raspberry Pi. The Photobooth itself could be only connected as client to it while running e.g. on a mini PC without GPIO.

Maybe this could be combined?

You're always welcome to fork the source and add new features or fix bugs. Your changes can be submitted for review while opening a Pullrequest.

Best regards

Andi

andi34, 2023-10-04:
https://github.com/frogro/PhotoboothProject_Pico_as_HID_Button_and_rotary_encoder

https://github.com/frogro/PhotoboothProject_Pico_W_as_remote_button_and_rotary_encoder

closed 2023-12-18 by stale[bot]