PhotoboothProject/photobooth #276: [Info]: Photobooth with PHP7

andi34, 2023-08-02:
### Is your feature request related to a problem?

Yes

### Description

```php
<?php
//
// source: Laravel Framework
// https://github.com/laravel/framework/blob/8.x/src/Illuminate/Support/Str.php

if (!function_exists('str_starts_with')) {
    function str_starts_with($haystack, $needle) {
        return (string)$needle !== '' && strncmp($haystack, $needle, strlen($needle)) === 0;
    }
}

if (!function_exists('str_ends_with')) {
    function str_ends_with($haystack, $needle) {
        return $needle !== '' && substr($haystack, -strlen($needle)) === (string)$needle;
    }
}

if (!function_exists('str_contains')) {
    function str_contains($haystack, $needle) {
        return $needle !== '' && strpos($haystack, $needle) !== false;
    }
}
?>
```

### Describe the solution you'd like

.

### Describe alternatives you've considered

.

andi34, 2023-08-02:
Update Apache Webserver
```
sudo apt install apt-transport-https lsb-release ca-certificates 
software-properties-common -y
sudo add-apt-repository ppa:ondrej/php -y
sudo apt update
sudo apt install libapache2-mod-php8.2 -y
sudo a2enmod php8.2
sudo apt install php8.2-{zip,gd} -y
sudo service apache2 restart
sudo gpasswd -a www-data plugdev
sudo gpasswd -a www-data video
sudo gpasswd -a www-data www-data
```

mart-e, 2023-08-12:
Hi,

I see you have just changed with 644833da5d33b0b7d2aec04a874e8d0b54367974 but this does not seem to work with my latest install of bullseye on raspberry pi

```
### Photobooth needs some software to run.
### Installing NGINX Webserver...
Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances... Fait
Lecture des informations d'état... Fait      
E: Impossible de trouver le paquet php8.2
E: Impossible de trouver de paquet correspondant à l'expression rationnelle « php8.2 »
E: Impossible de trouver le paquet php8.2-fpm
E: Impossible de trouver de paquet correspondant à l'expression rationnelle « php8.2-fpm »
```
Sorry in French but it says "impossible to find package php8.2"

Reading the commit, I see it was supposed to add a new source but the file `/etc/apt/sources.list.d/php.list` is not created (not sure if failed before or skipped the `if`)

andi34, 2023-08-12:
Maybe missing some packages

Does it work if you run
```
sudo apt install apt-transport-https lsb-release ca-certificates software-properties-common -y
sudo wget -qO /etc/apt/trusted.gpg.d/php.gpg https://packages.sury.org/php/apt.gpg
echo "deb https://packages.sury.org/php/ $(lsb_release -sc) main" | sudo tee /etc/apt/sources.list.d/php.list
```

andi34, 2023-08-12:
https://github.com/PhotoboothProject/photobooth/commit/619dfae15ca2785cf3cb0da7b132728a71f22f59

Was called after webserver install, must be added earlier. Also made sure needed packages are installed.

Gambit359, 2023-10-06:
Hallo,

i tried to install on a rasberry pi (new and clean OS) with install-photobooth.sh. Here i got a problem with php-dom.
It says something like "for this package there ist no installation ..."  ("existiert kein installationskandidat").
I changed in the COMMON_PACKAGES section of install-photobooth.sh php-dom to php8.2-dom and it seems to work on this section. 
Then i got other errors, but maybe i can figure this out for myself.

andi34, 2023-10-06:
php-xml instead

andi34, 2025-01-02:
PHP 8.3 required now. Installer handles PHP also on update.

closed 2025-01-02 by andi34