PhotoboothProject/photobooth #279: [FEATURE]: change background color of transition in collage

mart-e, 2023-08-17:
### Is your feature request related to a problem?

No

### Description

After taking a picture in a collage, we see the picture that was taken with a blue background.

[image]

I don't mind color but having a lighter, more neutral color would be probably be better.

### Describe the solution you'd like

Could be one of:

- Have a more neutral color (light blue, grey, black...) by default
- Be able to configure the background color (might be the case, sorry if I missed the option)
- Use the main screen background image

### Describe alternatives you've considered

Note that I am not sure if this is done by PhotoboothProject or by photobooth-app (that I am using the picamera)

andi34, 2023-08-17:
Hey and thanks for your request. Please check the UI section inside the adminpanel, the color is changeable already.

All elements can also be styled as needed with a custom css file (private/overrides.css), this also allows to use a background-image if needed.

mart-e, 2023-08-17:
Thanks for the quick reply !
I see, I was checking in the "Collage" settings but it is in the "User Interface" one.

However, I still wonder if the chosen color is correct.
I have tried changing the different settings and it seems that the one that the background color of the "Please wait" screen (as in the above picture) is the "Admin- and login panel color".
Is it expected?
[admin pannel](https://github.com/PhotoboothProject/photobooth/assets/564822/aab7931c-b4b9-48c4-8c9b-7f5ac1c8ebae)
[please wait](https://github.com/PhotoboothProject/photobooth/assets/564822/964d7bf8-3e5b-4acc-909b-98056ab583a4)

andi34, 2023-08-17:
I see, yeah we should change that. The question is what color would be best... Primary light? Countdown Background color?