PhotoboothProject/photobooth #310: Play a sound while the countdown is running

STarfield42, 2023-09-09:
### Is your feature request related to a problem?

No

### Description

Hello,

It would be cool if the countdown could also have an audio part.

### Describe the solution you'd like

A sound could be played for each tick of the countdown and a last one for the "cheeeeese" screen.

If we can have a bunch of different sounds, it would be less boring.
(I can provide some samples in french if you wish)

### Describe alternatives you've considered

I changed the source code for my  need.

I made 5 MP3 with several members of my family saying "5..4..3..2..1..Cheese!"

I added this code in the _core.js_ file, in the _api.startCountdown_ function definition (just before the definition of _timerFunction_) 

```
var soundfile = '/resources/snd/counter-' + Math.floor(Math.random()*6) + '.mp3';
var audio = new Audio(soundfile);
var dummy = audio.play().then(() => {}).catch((error) => {photoboothTools.console.log('Error with audio.play: ' + error);});
```

it works but it's a very basic implementation.

dadav, 2024-01-07:
I'd also love this. It would be also cool if we could set a sound for cheese (right before the photo is taken) too (and be able to only play this, without the countdown sounds).

m-anish, 2024-01-22:
I second this! Will be a great feature to have, and as part of the admin panel, so that the audio can play inside the browser, instead of some external app.

andi34, 2024-02-11:
Implemented with https://github.com/PhotoboothProject/photobooth/pull/587

closed 2024-02-11 by andi34