PhotoboothProject/photobooth #321: [FEATURE]: Take Picture from Android Smartphone via DroidCam + ADB

testadler, 2023-09-18:
### Is your feature request related to a problem?

Yes

### Description

How to use Smartphone Camera over ADB and Droidcam ? Upcycling old Smartphones for the Photobooth

### Describe the solution you'd like

Stream Live Picture and Take Pictures from the Android phone over adb connection. Controll the Phone via ADB Script.

### Describe alternatives you've considered

A Small Script get Startet if the take Photo Button get pressed. Starts a Script in the Background , Connect via ADB to the Phone and take ficture from the Camera. 

Also it is possible to use ffmpeg to stream the Live Kamera Feed from the Phone to the Website of the Photobooth

testadler, 2023-09-18:
Script for adb connect and droidcam control : 

#!/bin/sh
echo Beendende Laufende Apps
sudo adb shell am force-stop  com.dev47apps.droidcamx
sudo adb kill-server
echo Start ADB Server neu
echo Starte USB Modus
echo Starte Droidcam
sudo adb shell am start  com.dev47apps.droidcamx/com.dev47apps.droidcamx.DroidC>
echo Verbinde mit Droidcam Client
echo warte 10 Sekunden auf App
sleep 10
/opt/droidcam/droidcam-cli adb 4747 width=1920 height=1080

testadler, 2023-09-18:
Link to the Droidcam tool 
https://github.com/dev47apps/droidcam/wiki/Raspberry-PI

andi34, 2023-09-18:
Hey and thanks for your request. 
In general that's possible already: The take picture command and also the preview command can be defined as needed. There's also a pre- and post-picture command which can be defined as needed.

If you have batch file you can also use that to take an image. Make sure permissions are defined/ set property for the www-data user.

andi34, 2023-09-18:
In case your using fswebcam the take picture command is:
```
fswebcam --no-banner %s
```

closed 2023-09-18 by andi34