PhotoboothProject/photobooth #371: [BUG]: after taking picture white screen and gear is shown

peterwup, 2023-10-06:
### Describe the bug

- After taken the picture, only a white screen is shown and then a turning gear for > 60s.

- Even if the raspi has the right local time, picters in the galery a shown with UTC time stamp and not with local german time. 

### Steps to reproduce the behavior

Installed on a Raspberry 4b
older Raspberry Pi camera module is connected (for test purpose)
Raspberry pi is in my local home network by WIFI
Executed config reset in the Admin panel  (http://192.168.0.120/photobooth/admin/  from my windows computer)
Changed "Take picture command" to : `libcamera-still -n -o %s -q 100 -t 1 | echo Done`
Display connected to raspberry pi and in parallel a browser on my Windows computer
Display of the raspian shows photobooth in kiosk mode

- Taking a picture by touch screen on the raspberry pi 
- Taking a picture from the web browser of my PC

- Countdown is shown
- Picture is taken (it is in the gallery)
- white screen is shown
- gear is shown for a long time (> 60s):
[image]

### Expected behavior

Expected that the picture is shown fo a while after it has been taken.

### Photobooth Version

latest version

### Browser and browser version

Chrome

### Operating System

64bit Debian GNU/Linux 11 (bullseye) installed by Raspberry Pi Imager some days ago

### Did it work before?

No

peterwup, 2023-10-06:
libcamera-still  works: 

`libcamera-still -n -o demo.jpg -q 100 -t 1 | echo Done
Done
[0:57:19.498948325] [3384]  INFO Camera camera_manager.cpp:297 libcamera v0.0.5+83-bde9b04f
[0:57:19.536627813] [3386]  WARN RPI vc4.cpp:383 Mismatch between Unicam and CamHelper for embedded data usage!
[0:57:19.537506643] [3386]  INFO RPI vc4.cpp:437 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media4 and ISP device /dev/media2
[0:57:19.537586180] [3386]  INFO RPI pipeline_base.cpp:1101 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
[0:57:19.538492381] [3384]  INFO Camera camera.cpp:1033 configuring streams: (0) 1640x1232-YUV420
[0:57:19.539025360] [3386]  INFO RPI vc4.cpp:565 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 1640x1232-SBGGR10_1X10 - Selected unicam format: 1640x1232-pBAA
[0:57:19.892145223] [3384]  INFO Camera camera.cpp:1033 configuring streams: (0) 3280x2464-YUV420 (1) 3280x2464-SBGGR10_CSI2P
[0:57:19.893033831] [3386]  INFO RPI vc4.cpp:565 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 3280x2464-SBGGR10_1X10 - Selected unicam format: 3280x2464-pBAA
Still capture image received
`
Picture is created in the terminal as expected. 

And as I said it is available in the galery  if I take the picture by photobooth. This is what I am wondering. 
I will make a extra bug for the time stamp. 
thank you 
Peter

andi34, 2023-10-06:
Could you please update by hand once?

```
sudo apt install php-xml -y
sudo service apache2 restart
sudo -u www-data -s
cd /var/www/html
git fetch origin
git checkout origin/dev
git submodule update --init
npm install
npm run build
exit
```
Clear the browser Cache completely and try again.

closed 2023-10-06 by peterwup

andi34, 2023-10-06:
There have been some adjustments made , currently Photobooth is under heavy development. It's the manual way to update to latest development version.
php-xml is a new requirement and will be installed automatically on a clean Installation.

Great to hear it works! Feel free to join our Telegram group if you have questions or run into issues.