PhotoboothProject/photobooth #372: [BUG]: Timstamp shown in the gallery does not fit to the time when it is taken

peterwup, 2023-10-06:
### Describe the bug

The time stamp shown in the gallery for each picture is 2h behind the time where the pictures are taken.

### Steps to reproduce the behavior

Check date/time in the terminal
`date Fr 6. Okt 13:08:21 CEST 2023`

- take a picture by photobooth 
- select the gallery picture is taken at 13:12, but 11:12 is shown
[image]

- check the picture name/timestamp:
[image]

- picture has the right timestamp
- picture name seems to have UTC time which seems to be used by the gallery

### Expected behavior

In the gallery the pictures show the time when they are taken. 

### Photobooth Version

latest version

### Browser and browser version

Google Chrome Version 117.0.5938.132 (Offizieller Build) (64-Bit)

### Operating System

Raspberry Pi 64bit Debian GNU/Linux 11 (bullseye)

### Did it work before?

No

andi34, 2023-10-06:
https://stackoverflow.com/a/32227023

Maybe @benjaminkott has an idea how to make this available via Photobooth

peterwup, 2023-10-06:
Yes this solves the problem. 
@andi34  Thank you for the quick support. 

What I noted down for me and what maybe could be added to the FAQ:

**Timestamp of picture name and shown in gallery do not have the right time (wrong time zone) :**

Edit the following ini files for your PHP version (in this case 8.2):  (check with `ls /etc/php`  the available versions)

/etc/php/_8.2_/cli/php.ini
/etc/php/_8.2_/apache2/php.ini

sudo nano /etc/php/_8.2_/cli/php.ini
search in both files for the line: 

`;date.timezone` 

change to your time zone (here example for Germany): 

`date.timezone = Europe/Berlin`

restart the Raspberry

closed 2023-10-06 by peterwup