PhotoboothProject/photobooth #391: [BUG]: install script gives fatal error when changing 'html'

Notch001, 2023-10-13:
### Describe the bug

Hello, I received an fatal Error several times when running the installation script. If the script makes changes to the HTML folder, it gives the error "fatal error permission denied"I tried with several versions and also set the permissions for the folder manually beforehand. Unfortunately no success. The last attempt was with Raspberry PI OS  Desktop on a Raspberry Pi 4 64 bit

### Steps to reproduce the behavior

Simply run the installation script 

### Expected behavior

To install the software 

### Photobooth Version

Latest, link from wget in instructions 

### Operating System

Raspberry PI OS with Desktop 

### Did it work before?

No

andi34, 2023-10-14:
Hey and thanks for the report. I hope to get needed fixes in today..
Maybe you can try and confirm it's working using https://raw.githubusercontent.com/benjaminkott/photobooth/task/install-action/install-photobooth.sh

Notch001, 2023-10-14:
Hello, thank you for your answer. Unfortunately, I'm not quite fit enough
to understand what I should do now. Is this another installation script?

andi34, 2023-10-14:
Updated version of the installer, yes

```
rm install-photobooth.sh
wget https://raw.githubusercontent.com/benjaminkott/photobooth/task/install-action/install-photobooth.sh
sudo bash install-photobooth.sh -username='<YourUsername>'
```

It should contain everything needed to fix the permission problem.

Notch001, 2023-10-14:
I ran your new script on Raspberry Pi Is Lite 64. Unfortunately, there is
another error in npm. I have attached the error and part of the log file

deluxemongo, 2023-10-14:
On Raspberry PI OS bookworm i get the same error with npm. I think the package from "apt" is the problem. When modifying your install script to always use the function "update_nodejs()" a newer version of npm will be installed with the updater from stanley.

andi34, 2023-10-14:
> On Raspberry PI OS bookworm i get the same error with npm. I think the package from "apt" is the problem. When modifying your install script to always use the function "update_nodejs()" a newer version of npm will be installed with the updater from stanley.

I am aware of and Nodejs via stanley is removed on the updated script I posted.  

@Notch001 nothing attached, use the Webinterface.

deluxemongo, 2023-10-14:
> 
> I am aware of and Nodejs via stanley is removed on the updated script I posted.
> 

Does not work on a fresh install (of bookworm) with the updated script. Installed Nodejs version from apt sources matches 18.17, so the photobooth install script never runs "update_nodejs()" and installer fails with:

```
Cloning into '/var/www/html/vendor/Seriously'...
Submodule path 'vendor/Seriously': checked out '513d2b8696b76fe764b3033aa6dc7c99005acccd'
### Get yourself a hot beverage. The following step can take up to 15 minutes.
npm ERR! code EBADENGINE
npm ERR! engine Unsupported engine
npm ERR! engine Not compatible with your version of node/npm: photobooth@4.3.1
npm ERR! notsup Not compatible with your version of node/npm: photobooth@4.3.1
npm ERR! notsup Required: {"node":">=18.17.0","npm":">=9.6.0"}
npm ERR! notsup Actual:   {"npm":"9.2.0"}
```

Maybe a function to check npm version could be implementd, something like 

if $(npm -version) -le 9.6.0 -> run your updated part of nodejs update

I ran that part (update_nodejs())) manually and rerun the photobooth installer, I get the following error and setup breaks:

```
[Package]   Installing missing common package: npm
E: Unable to correct problems, you have held broken packages.

```

For now I removed 'npm' from the extra packages, the installer runs until the end and photobooth is installed.

Notch001, 2023-10-14:
Can you maybe tell me where you had to remove 'npm'?

deluxemongo, 2023-10-14:
> Can you maybe tell me where you had to remove 'npm'? 

For my described case, (install fails, manually update npm, rerun installer) I commented this line:

EXTRA_PACKAGES+=('npm')

But this is not the best way. If you are not familiar I think you should wait for the devolpers until the fix the problem within the installer. 

Aside from that this case is about permission errors on html folder and not a wrong npm version (on bookworm). The original posted bug seems to be fixed.

andi34, 2023-10-16:
Installer should be fixed now. If possible please do a fresh install of PiOS and try again.

closed 2023-10-16 by andi34

Notch001, 2023-10-16:
Yes, I did a fresh install and everything worked fine. Thanks

andi34, 2023-10-16:
Thanks for testing. 👍😊
There was a general issues with the permission after switching some installation steps to run as `www-data` user,.also there have been 2 issues on bookwork:
- Node.js was installed in v18.13 by default, npm was missing
- we switched to use the official node sources instead the pi binaries from audstanley

Now we also check for the minor version string of Node.js and suggest an update if needed.
If npm is missing we also try to install, npm version will be proofed also now and tried to be updated if it's not matching out requirements.