PhotoboothProject/photobooth #395: [BUG]: gallery.php freezes after viewing an image, requires refresh

mikerodrigues, 2023-10-16:
### Describe the bug

I'm using stable4 branch on Ubuntu with Nginx.

When I go to the standalone gallery at /gallery.php, I'm initially able to scroll, click the in-page refresh button (and have it refresh), and click into a gallery image to print it or do whatever.

However, after exiting that initial image, the gallery UI is completely frozen. Scrolling doesn't work, clicking on other thumbnails to open photos doesn't work, in-page refresh button doesn't work. Once I do a browser refresh of the page, it works, but only for one instance of "focusing on an image by clicking on a thumbnail".

I can click on a thumbnail, and then use the left/right arrows to flip between images in the gallery, but once I close it, the page is totally frozen.

I've tried this on Firefox, Chrome both on Linux and Android on two different devices, and I get the same behavior everywhere. I've disabled adblockers.

I do notice that in the FAQ, the URL is "/gallery". In practice, I have to go to "/gallery.php" with the explicit ".php" for it to load anything. "/gallery" gives me a 404.

nginx config:
```
server {
	listen 80 default_server;
	listen [::]:80 default_server;
	root /var/www/html/photobooth;
	index index.php index.html index.htm index.nginx-debian.html;
	server_name _;
	location / {
		try_files $uri $uri/ =404;
	}
	location ~ \.php$ {
		include snippets/fastcgi-php.conf;
		fastcgi_pass unix:/var/run/php/php8.1-fpm.sock;
	}
}
```

### Steps to reproduce the behavior

1. Go to https://localhost/gallery.php
2. Click any thumbnail
3. Close out of the picture 
4. Try to scroll or click anything on the page, it won't be responsive

### Expected behavior

1. Go to https://localhost/gallery.php
2. Click any thumbnail
3. Close out of the picture 
4. Click on another picture, or scroll, or click the refresh button on the page, and have it respond by opening that picture, scrolling the page, or refreshing the gallery.

### Photobooth Version

stable4

### Browser and browser version

Firefox 118 Linux & Android Chrome 117 Linux & Android

### Operating System

Ubuntu 22.04 & Android

### Did it work before?

No

andi34, 2023-10-16:
Hey and thanks for the report. Please update to latest development version and check if the issue still exist and reopen the issue as needed. Issues on latest stable release won't be looked at anymore in favor of the upcoming Photobooth V5.

mikerodrigues, 2023-10-27:
FWIW, the php 8.2 requirement for development makes it too difficult for me to use on 22.04, which only has 8.1 in the official repos. Idk what current Debian is running. I imagine most people would want to run this on LTS builds to avoid maintenance.

andi34, 2023-10-27:
v4 reached its EOL and all known bugs have been fixed in the meantime. We're hard working on finishing our work for the next version.
It's quite easy to update PHP on Ubuntu.
```
sudo add-apt-repository ppa:ondrej/php -y
sudo apt update

sudo apt install php8.2 php8.2-cli libapache2-mod-php8.2 -y
sudo apt install php8.2-{zip,gd,xml,mbstring} -y
sudo a2dismod php8.1
sudo a2enmod php8.2
sudo service apache2 restart
```

Just to make sure permissions matches you should also run the following :
```
sudo gpasswd -a www-data plugdev
sudo gpasswd -a www-data video
sudo gpasswd -a www-data www-data
sudo reboot
```

andi34, 2023-10-27:
About http://localhost/gallery.php vs http://localhost/gallery, that was changed on 15. June. The gallery.php was moved for better maintainability.

andi34, 2023-10-27:
I'll try to update latest V4 again with some Bugfixes and small improvements from latest development branch.
It'll still require node.js in v18.17 and npm instead of yarn.
The goal is to retain compatibility with latest Photobooth installer and fixing the known bugs of Photobooth V4. Small code improvements have also been applied and a handful of small features will be added.

https://github.com/PhotoboothProject/photobooth/tree/update/stable4

It's still WIP and needs to be tested, especially the remotebuzzer server.

andi34, 2023-11-24:
Issue can be reopened if still is a problem on latest development version.

closed 2023-11-24 by andi34