PhotoboothProject/photobooth #414: [BUG]:  SYNCH to USB not working

RickCopithorne, 2023-10-23:
### Describe the bug

Enabling the Synch to USB is not working. Once enabled within the ADMIN console and setting the Config to look for either /dev/sda or the Removable USB Name the files are never synched to the USB drive. 

When looking at the Debug panel I see the following
sh: 1: /usr/bin/node: not found

### Steps to reproduce the behavior

goto http://localhost/admin/
Click Synch Pictures to USB Stick
Under USB Identifier add the location /media/pbadmin/photobooth

### Expected behavior

After the allocated time the new photos should be copied from the Photobooth to the USB stick, 

### Photobooth Version

Latest Version

### Browser and browser version

Chromium 116.0.5845.102

### Operating System

Raspbian Bullseye v11

### Did it work before?

No

andi34, 2023-10-23:
`sh: 1: /usr/bin/node: not found` means you need to fix the missing library

andi34, 2023-10-23:
Node path can be defined via Adminpanel.

RickCopithorne, 2023-10-23:
Will that break anything else ? And this was just a basic install from the documents in Git so what’s the proper path to use ?

andi34, 2023-10-24:
What's the output of `sudo -u www-data which node` ?

RickCopithorne, 2023-10-24:
Nothing happens. Once I hit enter it brings me to a new prompt

***@***.***:~ $ sudo -u www-data which node
***@***.***:~ $

Rick Copithorne

Principal Systems Administrator

work: +1 781 221 6808  |  www.aspentech.com<http://www.aspentech.com>

andi34, 2023-10-24:
So node is not available, at least not for the www-data user which is running the webserver and running the sync script. You need to get that fixed.

RickCopithorne, 2023-10-24:
Hate to ask any suggestions? I did enable USB synch during the install and also ran the enable-usb-aynch scripts. ( guessing the name .

andi34, 2023-10-24:
[image]

Your setup must look like this. There's something wrong with your installation.

I'd suggest you try a clean install using latest installer.

andi34, 2023-10-24:
Or get it fixed while installing node.js in v18

closed 2023-10-24 by andi34