PhotoboothProject/photobooth #503: [BUG]: Photo not saved - command line works though

SvHeMobile, 2023-12-16:
### Describe the bug

Hi,

when I use command line --capture-image-and-download everything works fine, also with www-data user. However, I can't get it to work via the application. 
Camera is triggered, picture is taken but then the error message comes. 
Collection works up to final picture (I also see al photos). Thus, must be a permission issue or so.

Raspberry Pi 3B, latest OS (minimal)
Latest version of Photobooth
Canon EOS 600D

### Steps to reproduce the behavior

-

### Expected behavior

-

### Photobooth Version

4.3.1

### Browser and browser version

Chrome

### Operating System

Raspberry Pi OS

### Did it work before?

No

### Latest working version

-

### Additional context

-

andi34, 2023-12-18:
You've used the installer for setup or setup by hand?
Followed the instructions inside the FAQ?
https://photoboothproject.github.io/FAQ#i-get-the-error-message-something-went-wrong-while-taking-a-picure-what-can-i-do

Please post the Photobooth log, access via Adminpanel -> Debugpanel

Also please post the output of
```
cat /var/www/html/HEAD
```
(Enter in your terminal)

SvHeMobile, 2023-12-19:
Hi, I'm the problem it's me... [contemporary American philosopher]
After hours of trying to find the glitch, I realized I picked the only ancient Raspberry 2B I had in my drawer. After discovering that and switching to a 3B, it works perfectly fine.
Thanks for the effort you put into this project!
And sorry for the "fake" bug...

andi34, 2023-12-19:
Thanks for the feedback. Maybe a Pi Camera works on a Pi 2, in case it's an option for you too.

Closing the issue. If you have further questions you're also welcome to join our community on telegram.

Best regards

Andi

closed 2023-12-19 by andi34