PhotoboothProject/photobooth #524: [BUG]: Request-URI Too Long

dadav, 2024-01-02:
### Describe the bug

It's not possible to change settings via http://localhost/admin because the resulting http-get request is too long (code 414).

### Steps to reproduce the behavior

```bash
git clone git@github.com:PhotoboothProject/photobooth.git
cd photobooth
docker-compose up --build
# now open http://localhost/admin and try to change the settings..
```

### Expected behavior

Clicking on save should not result in an error.

### Photobooth Version

b3df99abd9fed628676c27974e6e0646a6ae97be

### Browser and browser version

Firefox 121.0

### Operating System

Archlinux

### Did it work before?

No

dadav, 2024-01-02:
The default maximum length [seems to be 8190](https://httpd.apache.org/docs/2.2/mod/core.html#limitrequestline) and the request is >10.000 chars long.

closed 2024-01-09 by andi34

wardaddyuno, 2024-01-14:
How to fix this error on windows? I have the settings page loaded with no stylesheet or scripts loaded.

andi34, 2024-01-14:
Edit `C:\Apache24\conf\httpd.conf` and add
```
LimitRequestLine 12000
```