PhotoboothProject/photobooth #526: [FEATURE]: Support live preview in windows

dadav, 2024-01-03:
### Is your feature request related to a problem?

Yes

### Description

Currently the `cameracontrol.py` script relies on `python-gphoto2` which is linux/macos only. It would be cool to have an windows solution too. The digicamcontrol project has the possibility to start a webserver, start the live view and provide access to the stream via `http://localhost:5513/live`...maybe this can be wrapped in a script.

### Describe the solution you'd like

A script which makes it possible to have live preview of the dslr. Something like this:

```ps
# start the live view (hidden)
Start-Process -WindowStyle hidden -FilePath CameraControlRemoteCmd.exe -ArgumentList "/c do LiveViewWnd_Show"

# During sleep, access stream at http://localhost:5513/live
Start-Sleep -Seconds 5

# cancel live view
Start-Process -WindowStyle hidden -FilePath CameraControlRemoteCmd.exe -ArgumentList "/c do LiveViewWnd_Hide"
```

### Describe alternatives you've considered

-

andi34, 2024-01-03:
Hey and thanks for the report/request.
In General, preview is possible on all OS - but only for Linux there are known ways using gphoto2/libgphoto2 while creating a virtual webcam.

Photobooth uses MediaStreamConstraints to use a webcam/virtual webcam for preview. A different way is to use the preview via URL using CSS.

For the start preview command we only use the OS command line, this could also be a batch file  you call on Windows (for Windows small adjustments might be needed inside api/previewCamera.php if validation is active).

The preview command is executed independent of the preview mode.

In short: If you're able to start the preview via batch file you're able to use the preview from URL `url(http://localhost:5513/live)`

andi34, 2024-01-03:
(you could also try the pre-photo command if start Preview command makes trouble)

dadav, 2024-01-03:
> Hey and thanks for the report/request. In General, preview is possible on all OS - but only for Linux there are known ways using gphoto2/libgphoto2 while creating a virtual webcam.
> 
> Photobooth uses MediaStreamConstraints to use a webcam/virtual webcam for preview _"from device cam"_. A different way is to use the preview _"from URL"_ using CSS.
> 
> For the start preview command we only use the OS command line, this could also be a batch file you call on Windows (for Windows small adjustments might be needed inside api/previewCamera.php if validation is active).
> 
> The preview command is executed independent of the preview mode.
> 
> In short: If you're able to start the preview via batch file you're able to use the preview _"from URL"_ like this: `url(http://localhost:5513/live)`

Thanks for your reply. Setting the css is a good idea. I'll also have a look in that php file, I think my windows installation is missing pgrep which is used by that script. I also noticed that there is probably a conflict with the running digicamcontrol instance, which is required for the preview, and the capture command, which also wants to access the camera. The digicamcontrol instance needs some startup time, so maybe I'll start it AFTER the photo was taken and kill it before raking the photo.. I'll report back later.

andi34, 2024-01-05:
Yeah, the pgrep would be an issue, but shouldn't be executed if _Execute preview commands without validation_ is enabled (default). Maybe you can call another batch file to kill digiCamControl via stop command?

andi34, 2024-01-05:
And for capture another batch file which will capture 
`http://localhost:5513/?CMD=Capture` and download the image `http://localhost:5513/preview.jpg`, destination need to be adjusted and is given by Photobooth via `%s` and can be passed.

dadav, 2024-01-05:
> And for capture another batch file which will capture `http://localhost:5513/?CMD=Capture` and download the image `http://localhost:5513/preview.jpg`, destination need to be adjusted and is given by Photobooth via `%s` and can be passed.
> 
> See https://www.digicamcontrol.com/doc/userguide/web also for adjusting the default capture folder etc.

Thanks for the tips. I wrote [this simple wrapper](https://github.com/dadav/digipy/blob/main/digipy.py) which starts the Liveview, captures an image and kills the Liveview. Only the `url(http://localhost:5514/live)` setting is not working yet. I need to investigate further..

The script can be used like this, if anyone is interested:

```bash
# start Liveview
python digipy.py cmd LiveViewWnd_Show

# Capture a photo
python digipy.py slc capture %s

# Cancel Liveview
python digipy.py cmd LiveViewWnd_Hide
```

andi34, 2024-01-06:
Great! Thanks a lot for taking this challenge!
Because of the typo 5513 vs 5514?
Can you access the URL independent of Photobooth to see the preview?

About your repo:
Small readme with instructions on how to setup step by step? (Most user might already stuck while installing python)
License?

dadav, 2024-01-06:
> Great! Thanks a lot for taking this challenge! Because of the typo 5513 vs 5514? Can you access the URL independent of Photobooth to see the preview?
> 
> About your repo: Small readme with instructions on how to setup step by step? (Most user might already stuck while installing python) License?

The typo was one problem, yrs. 5514 is correct (default in digicamcontrol for the stream). I'm able to view it in the browser, but it crashes sometimes. There are issues in their repo about that, but it's still unfixed.

I'm still working on the digipy wrapper repo and will add everything needed.

dadav, 2024-01-09:
Ok, I found a way better method using the Linux subsystem for windows. I can now use gphoto2 with windows 👍🏻 I will now try to get the preview script working and write everything I did together.

andi34, 2024-01-09:
Great!
To be true, Ive heard it works on wsl but haven't had the chance to test. This would make installation for windows users much easier I think.

I asked Benjamin also to write the steps down to install Photobooth from source on Windows too. This way we wouldn't depend on the release packages.

dadav, 2024-01-09:
> Great! To be true, Ive heard it works on wsl but haven't had the chance to test. This would make installation for windows users much easier I think.
> 
> I asked Benjamin also to write the steps down to install Photobooth from source on Windows too. This way we wouldn't depend on the release packages.

Where I need to send the camera stream to? I've tried to read the cameracontrol.py script, which seems to send the data via ffmpeg to some device file. I think I need to find another way to get the data out of the wsl box..

andi34, 2024-01-10:
Possible to make it an stream available via http?
Maybe parts of  https://github.com/photobooth-app/photobooth-app/tree/main/photobooth/services/backends can help?

dadav, 2024-01-10:
> Possible to make it an stream available via http? Maybe parts of https://github.com/photobooth-app/photobooth-app/tree/main/photobooth/services/backends can help?

I think I could make it work with go2rtc. Will try and give feedback :)

dadav, 2024-01-10:
Almost there....https://github.com/AlexxIT/go2rtc/pull/871

dadav, 2024-01-11:
I can't seem to execute the capture command:

```txt
[cmd] => wsl gphoto2 --capture-image-and-download-filename=C:\Apache24\htdocs\data\tmp\20240111_212300.jpg 2>&1
```

Fails with `Permission denied`. But I don't know why.

Another problem is that the connection to the stream stays alive and so the capture-movie commands keeps running. Is it possible to close the preview connection before taking executing the capture command?

andi34, 2024-01-12:
On Linux we use the cameracontrol.py to start the stream, if that's used we also need to capture via cameracontrol.py
Else you can define a stop command which is executed.

andi34, 2024-01-12:
There's also a docker file, not sure if the USB connected camera can be used there?

dadav, 2024-01-12:
> There's also a docker file, not sure if the USB connected camera can be used there?

Hm, might worth a try. It's certainly possible in linux, not so sure in windows tho.

dadav, 2024-01-12:
Currently trying to run a python script as the capture command. This is driving me insane. The error tells me it can't execute python3.exe or the script (not sure), but everything seems correct.

closed 2024-01-20 by dadav