PhotoboothProject/photobooth #527: [BUG]: Canon EOS 60D on Raspberry Pi 5 Liveview not working

r07aNd, 2024-01-03:
### Describe the bug

Hello,

first of all thank you for the great work. I installed a brand new Rasperry Pi 5 and installed Photobooth with the installation script. I am working directly on the Raspberry with a display. I changed the following settings in admin panel to enable preview with Canon EOS 60D:

Live preview -> Preview mode
	from: None
	to: from device cam (no remote preview)

Live preview -> Use preview from device cam as background
	from: Off
	to: On

Commands -> Execute start command for preview on take picture/collage
	from: On (default)
	to: Off

Commands -> Command to generate a live preview
	from: <empty>
	to: python3 cameracontrol.py
	
Commands -> Take picture command
	from: gphoto2 --capture-image-and-download --filename=%s
	to: python3 cameracontrol.py --capture-image-and-download %s

The following commands are working standalone in terminal:
* python3 cameracontrol.py --capture-image-and-download %s (takes a picture as expected)
* python3 cameracontrol.py (I can see the camera stream if I connect VLC to capture device /dev/video0)

If I open Photobooth via http://localhost/ the shutter is opening, but I only see the standard background of Photobooth. If I press the picture button to take a photo, the counter starts ending with the Cheeeeese message (stucks there without taking a picture). If I want to open http://localhost/ in a second browser window, Photobooth is not responding anymore.

Is there something else I could do?

Thank you!

### Steps to reproduce the behavior

see above

### Expected behavior

see above

### Photobooth Version

latest version (installed on 03.01.2024 via install-photobooth.sh)

### Browser and browser version

Firefox for Raspberry Pi 121.0

### Operating System

Raspberry Pi OS with desktop 64-bit (Bookworm)

### Did it work before?

No

andi34, 2024-01-04:
Hey and thanks for the report!

I am aware of the issue, but yet didn't hat enough time to look at it.
To workaround that please add the following lines to `/usr/ffmpeg-webcam.sh`
```
sleep 5
python3 /var/www/html/api/cameracontrol.py -b
```
(The `-b` is optional, remove if you set `Execute start command for preview on take picture/collage` to off)

r07aNd, 2024-01-04:
Hi,

thank you for the hint, but unfortunately it is not solving the problem. When I now start the Raspberry without accessing the the Photobooth application the shutter opens already (assuming this is because the service is started already?). When I open Firefox at http://localhost a small icon appears informing that a video is played in Picture-in-Picture mode. When I press on this small icon I only see black screen any video stream from the camera. When I press the picture button the countdown is triggered and a picture is taken. Liveview is unfortunately still not working.

kr

andi34, 2024-01-04:
What's your preview config? 
Same on Chromium?

This workaround is working for others. Any other cameras connected?

Maybe also try without using it as background using `-b` and starting preview on take picture/collage.

r07aNd, 2024-01-04:
Hi,

I just tested with Chromium and it works with Chromium - THANK YOU!

Besides the standard / default configuration after installation with the install script, I did the following changes:

Live preview -> Preview mode
	from: None
	to: from device cam (no remote preview)

Live preview -> Use preview from device cam as background
	from: Off
	to: On

Commands -> Execute start command for preview on take picture/collage
	from: On (default)
	to: Off

Commands -> Command to generate a live preview
	from: <empty>
	to: python3 cameracontrol.py
	
Commands -> Take picture command
	from: gphoto2 --capture-image-and-download --filename=%s
	to: python3 cameracontrol.py --capture-image-and-download %s

As workaround add the following lines to /usr/ffmpeg-webcam.sh (works only for Chromium)
	sleep 5
	python3 /var/www/html/api/cameracontrol.py

I have no other cameras connected.

For Firefox it does not work - here some screenshots maybe they will help to analyze further.

[image]

After pressing the small picture-in-picture icon it looks like this with Firefox

[image]

kr

andi34, 2024-01-04:
Access was granted inside Firefox?

r07aNd, 2024-01-04:
Yes permissions for camera usage has been granted within Firefox

andi34, 2024-01-05:
https://support.mozilla.org/en-US/kb/turn-picture-picture-mode

Could you please disable PIP and try again?

r07aNd, 2024-01-05:
I have now reset all firefox settings and tried again without success and have reset again all settings and disabled PIP with the same result for Mozilla Firefox 121.0 (64-bit) for Raspberry Pi OS, which was part of the installation image of the Raspberry PI.

I installed now in addition Firefox ESR (Extended Support Release) version 115.6.0esr (64-bit) **and there it works without problems** (with and without PIP)

Thank you very much for your support!

kr

SvenGuthe, 2024-04-05:
I have the same issue like described above.

When I execute `python3 cameracontrol.py` I clearly see that the video works. I also hear some sounds from the camera. But when I activate the livepreview and using `from device cam (no remote preview)` and turning `Execute start command for preview on take picture/collage` off, there is wheter a sound on my camera nor a live preview when taking a picture.

timo619, 2024-04-19:
I can still replicate the problem, that the first image is not taken. That is what I tested, hopefully it helps discovering the problem:
- my setup: Ubuntu 22.04, DELL thin client, Canon 5D MkII, with live view from device during count down
- when I press the "take picture" button I can see the live view and after count down the photobooth website keeps showing an error message, that it will restart, but it doesn't
- I am also unable to access admin page with the 6 clicks in the right buttom corner on first run, only possible after taking first picture and restarting firefox
My testing:
- Firefox 116 (snap) --> not ok
- Firefox ESR 102 (snap) --> not ok
- Chromium --> not ok
- disabled live view --> ok, everything works as expected (= firefox starts automatically in kiosk mode and I am able to take pictures)

andi34, 2024-04-20:
Please use the mjpeg/go2rtc variant for preview via DSLR until the issue is solved.

SvenGuthe, 2024-04-20:
I use mjpeg now. But there are also issues. Preview via Background url (CSS) does Not Work for me (CORB Issues) and the Default URL was wrong. I scripted a small workaround with an IFrame - what works perfectly. But thanks for all the hints - helped me a lot.

timo619, 2024-05-01:
With the updated version (01. May 2024) and mjpeg/go2rtc the issue (no first photo after start/reboot) is resolved, at least for me. 

**My setup again:**
- Ubuntu 22.04, DELL thin client, Canon 5D MkII, with live view from device during count down
- HDMI Touch display
- Firefox (snap) with autostart and kiosk mode
- use of mjpeg/go2rtc
- running localhost

Thank you very much for help and your work. This is a really nice project.

closed 2024-07-14 by stale[bot]