PhotoboothProject/photobooth #528: [BUG]: Remotebuzzer Server with GPIO on Raspberry Pi 5 not working

andi34, 2024-01-04:
### Describe the bug

Currently the Remotebuzzer Server with GPIO on a Raspberry Pi 5 is not supported.

### Steps to reproduce the behavior

Start the Remotebuzzer Server and enable hardware button support.

### Expected behavior

.

### Photobooth Version

Latest

### Operating System

PiOS bookworm

### Did it work before?

No

### Latest working version

None

### Additional context

The syfs interface to control the GPIO is different on a raspberry pi 5. See https://forums.raspberrypi.com/viewtopic.php?t=359302

The onoff library uses the sysfs interface, which might get deprecated for all Raspberry Pi.

```
GPIO   2 = gpio-460 (PIN3)
GPIO   3 = gpio-461 (PIN5)
GPIO   4 = gpio-462 (PIN7)
GPIO 14 = gpio-472 (PIN8)
GPIO 15 = gpio-473 (PIN10)
GPIO 17 = gpio-475 (PIN11)
GPIO 18 = gpio-476 (PIN12)
GPIO 27 = gpio-485 (PIN13)
GPIO 22 = gpio-480 (PIN15)
GPIO 23 = gpio-481 (PIN16)
GPIO 24 = gpio-482 (PIN18)
GPIO 10 = gpio-468 (PIN19)
GPIO   9 = gpio-467 (PIN21)
GPIO 25 = gpio-483 (PIN22)
GPIO 11 = gpio-469 (PIN23)
GPIO   8 = gpio-466 (PIN24)
GPIO   7 = gpio-465 (PIN26)
GPIO   0 = gpio-458 (ID_SD)
GPIO   1 = gpio-459 (ID_SC)
GPIO   5 = gpio-463 (PIN29)
GPIO   6 = gpio-464 (PIN31)
GPIO 12 = gpio-470 (PIN32)
GPIO 13 = gpio-471 (PIN33)
GPIO 19 = gpio-477 (PIN35)
GPIO 16 = gpio-474 (PIN36)
GPIO 26 = gpio-484 (PIN37)
GPIO 20 = gpio-478 (PIN38)
GPIO 21 = gpio-479 (PIN40)
```

maxlin1, 2024-01-04:
Here is the Pin Overview: https://community-storage.element14.com/communityserver-components-secureimagefileviewer/telligent/evolution/components/attachments/13/153/00/00/00/14/83/85/GPIO-Pi5v14.png-2080x1244.png?sv=2016-05-31&sr=b&sig=gGhGyYCOO%2FWsyltcEkWawjADIRqSrrv3HJdZBbLTyzU%3D&se=2024-01-10T23%3A59%3A59Z&sp=r&_=IyMgkX+9UuZLmCs//RUL0w==

andi34, 2024-01-04:
Doesn't help, see post above. Pi5 uses high-value GPIOs on sysfs

andi34, 2024-01-08:
Instructions have been added to the FAQ.
Installer needs to be adjusted (detect Pi5 and set config.txt accordingly).

How to handle default GPIO setup? Add a pre-defined Pi 5 my.config.inc.php on install? (After resetting it'll be lost).

fmalcher, 2024-01-09:
Thanks for pointing out the differences! What would be the current workaround to get remotebuzzer working on RPI5?

- I added the `gpio` config in `config.txt` for the `4xx` pins (as written in the FAQ and your note above).
- I changed the GPIO numbers in the photobooth config to `4xx` according to your mapping table. `4xx` numbers are visible in the combined config in the Debug Panel.
- Buttons are wired against GND. The whole setup worked fine with a RPI4.

remotebuzzer log does not show any errors:

```
[2024-01-09 13:18:52][remotebuzzer][DEBUG] 1142: BUTTON support active
[2024-01-09 13:18:52][remotebuzzer][DEBUG] 1142: Looking for Picture Button on Raspberry GPIO 420
[2024-01-09 13:18:52][remotebuzzer][DEBUG] 1142: Looking for Collage Button on Raspberry GPIO 419
[2024-01-09 13:18:52][remotebuzzer][DEBUG] 1142: Looking for Shutdown Button on Raspberry GPIO 415
[2024-01-09 13:18:52][remotebuzzer][DEBUG] 1142: Looking for Print Button on Raspberry GPIO 425
[2024-01-09 13:18:52][remotebuzzer][DEBUG] 1142: LED for Print Button on Raspberry GPIO 409
[2024-01-09 13:18:52][remotebuzzer][DEBUG] 1142: Initialization completed
[2024-01-09 13:18:52][remotebuzzer][DEBUG] 1142: socket.io server started
[2024-01-09 13:18:52][remotebuzzer][DEBUG] 1142: New client connected - ID Rtompi1cPGfRChb3AAAB
```

Is there anything else I need to do now?
Thanks!

>  Installer needs to be adjusted (detect Pi5 and set config.txt accordingly).

Isn't it possible to use the config.txt filters for this?
Something like this:

```
[all]
gpio=5,6,7,8,16,17,20,21,22,26,27=pu
gpio=9,10,11,12,18,19,23,24,25=op

[pi5]
gpio=404,405,406,407,415,416,419,420,421,425,426=pu
gpio=408,409,410,411,417,418,422,423,424=op

andi34, 2024-01-09:
Enable experimental settings view and make your adjustments inside the config too

fmalcher, 2024-01-09:
Thanks for the quick reply! Unfortunately, that doesn't seem to help.
Do you have any other ideas?

[image]

andi34, 2024-01-09:
Old Kernel?
Maybe run an apt update & apt upgrade

fmalcher, 2024-01-09:
It was a fresh new installation of bookworm 64-bit, so I suppose the kernel version is okay. `apt upgrade` and `apt dist-upgrade` had nothing to install. I can even switch on one of the LEDs through `/sys/class/gpio`, but the remotebuzzer server doesn't want to play with the GPIOs.

Do you have any other hint for me? Maybe I forgot something obvious?
Thanks a lot for your help!

andi34, 2024-01-09:
What's the output of
```
cat /sys/kernel/debug/gpio
```

About the config.txt I guess I'd do the config via detection of the Pi 5 by `$(tr -d '\0' </proc/device-tree/model)`. We could also use that inside PHP to define the default config.

andi34, 2024-01-09:
Maybe try a different kernel
```
sudo rpi-update rpi-6.1.y
```
Or
```
sudo rpi-update rpi-6.6.y
```

fmalcher, 2024-01-10:
Thank you! Unfortunately I didn't get it to work so I assembled it back to a RPI4.
Kernel 6.6 showed a whole different range of GPIO numbers (5xx). An update to the latest patch version of 6.1 didn't work either. `/sys/kernel/debug/gpio` showed the correctly assigned GPIO pins, however remotebuzzer didn't want to react to them.
On the RPI4 with the exact same SD card it worked out-of-the-box. So let's see how this evolves 🙂

andi34, 2024-01-10:
I need to setup my Pi5 still... But I was told it works. Could you test with one button connected to the GND only please?

maxlin1, 2024-01-10:
I ran my Pi 5 on Monday again completely according to the instructions on the website and the current PI OS (card reformatted).
It worked great except for one small thing. You have to do a #sudo apt update and sudo apt upgrade manually once. Otherwise you will currently get an npm error. B

andi34, 2024-01-10:
@maxlin1 why ever the NPM version check wasn't executed again or faulty, it still showed the initial version detected. I've added a recheck this morning slightly different.

It should be enough to run the installer a second time if the error occurs as the update of NPM worked.

I'd be glad if someone can test on a fresh install of pi os and report back.

andi34, 2024-01-10:
The `apt upgrade` might cause issues for those having the v4l2loopback installed already as it depends on the kernel. Not sure if there'll only be a problem on 32bit PiOS because of the 64bit Kernel used or also on 64bit OS.

Anyway:

You can also add apt update and apt upgrade to the installer by hand if needed.

andi34, 2024-01-12:
@maxlin1 in case you're able to test a clean I stall on a fresh PiOS Bookworm.

https://gist.github.com/andi34/c162e512aff4315306bfaf3abeac87df

Should set the overlay for Pi 5 inside config.txt and also default config inside Photobooth.

Hopefully really fixes the Problem updated npm not detected.

Both changes are ready for review and need to be tested.

andi34, 2024-01-12:
@fmalcher lol, most of the time it's detected and logged if it's recommended to add a pull-up resistor.  But glad it works! 

For testing I currently use 

AZDelivery 3 x KY-004 Taster Modul Sensor Taste Kopf Schalter Schlüsselschalter kompatibel mit Arduino und Raspberry Pi inklusive E-Book! https://amzn.eu/d/eydbGja

closed 2024-01-14 by andi34