PhotoboothProject/photobooth #562: [FEATURE]: Support different collage options from frontpage instead of admin

m-anish, 2024-01-22:
### Is your feature request related to a problem?

No

### Description

First of all. Wonderful project!

I was wondering if its possible to expose some of the collage layouts selectable through admin panel to the "user" from the frontpage

### Describe the solution you'd like

One of the closed source photobooth app we use supplies this feature, and we are considering switching to this software, so wondering if its possible to have this feature

### Describe alternatives you've considered

N.A.

andi34, 2024-01-22:
Hey and thanks for the request!
Currently that's not possible and the source would need adjustments in different places. Currently I've only limited time to work on this project and won't be able to add this feature soon, but we can keep this open for later implementation.
Besides that, of you're taking the challenge to implement it in a useful way I'd be glad to review your changes via Pull request.

m-anish, 2024-01-22:
Thanks for your reply.. i'd love to tinker at some point but my knowledge of js is quite limited. 

But great project! I had a lot of fun installing it on a rpi-4 and hopefully can make it into a photobooth one day :)