PhotoboothProject/photobooth #564: [FEATURE]: Remote DSLR preview using ffmpeg

DominicWindisch, 2024-01-22:
### Is your feature request related to a problem?

No

### Description

I plan on using an old android tablet as screen for the project running on a Pi 4B with a Canon camera connected via USB. 

Currently, a preview from the cam is only supported on localhost. 

### Describe the solution you'd like

Since it's using ffmpeg in the backend, it should in theory be possible to output the camera into a webrtc stream or alike which can then be accessed remotely. 

### Describe alternatives you've considered

I can't think of an alternative way using the existing hardware. I'd have to switch to a screen + keyboard.

### Additional context

The 3d printing project "octoprint" uses the same principle, streaming the camera live view in local network from a Pi works perfectly fine there. May be a starting point to get an idea on how to implement that...

dadav, 2024-01-22:
This is already possible if you add `--mjpeg` to the install script. It will go2rtc, which can stream the liveview even over internet.

DominicWindisch, 2024-01-22:
Oh wow, I totally missed that. To be fair, there is no description on the `-mjpeg` option in the install scrips help output. And it's not mentioned in the DSLR preview FAQs. I'll check if I get it to work. :)

DominicWindisch, 2024-01-22:
I've installed it. What's the command to initiate the RTC stream from Photobooth? It's not live yet at <IP>:8080/stream.

dadav, 2024-01-22:
Try this: http://localhost:1984/stream.html?src=dslr&mode=mjpeg

DominicWindisch, 2024-01-22:
That works! Thank you so much.

DominicWindisch, 2024-01-23:
I'm sorry to keep coming back to this issue, but I can't seem to get the preview stream to display in Photobooth. Am I missing a setting somewhere?

Accessing the stream directly works perfectly fine. I've set preview mode to "from url" and set the URL to the exact url working in the browser: "url(192.168.0.174:1989/stream.html?arc=DSLR&mode=mjpeg)".

I've changed the preview toggles to any thinkable combination but I can't seem to get a preview in Photobooth...

simoes-mickael, 2024-01-23:
Hello, same issue for me. Direct stream is ok but nothing in the photobooth app...

DominicWindisch, 2024-01-23:
I've tried both with and without http in the URL - no difference :(

DominicWindisch, 2024-01-23:
Oddly enough, using an MJPEG stream from a different device works. 

When changing the URL to my go2rtc instance on a different pi providing an MJPEG stream from my octopi cam, I can see the live stream on the Photobooth count down as expected. 

Maybe a go2rtc or ffmpeg configuration issue? Is there any config information that would help debugging?

dadav, 2024-01-23:
Please edit the file `/etc/go2rtc.yaml` and add this:

```yaml
log:
  level: debug
```

Then run `systemctl restart go2rtc`, do your testing and then run this: `journalctl -u go2rtc -b0 | tee go2rtc.log`  and upload the generated log file.

DominicWindisch, 2024-01-23:
Here's the output:

```
Jan 23 21:00:00 photobooth go2rtc[3018]: exit with signal: interrupt
Jan 23 21:00:00 photobooth systemd[1]: Stopping go2rtc.service - go2rtc streaming software...
Jan 23 21:00:00 photobooth systemd[1]: go2rtc.service: Deactivated successfully.
Jan 23 21:00:00 photobooth systemd[1]: Stopped go2rtc.service - go2rtc streaming software.Jan 23 21:00:00 photobooth systemd[1]: go2rtc.service: Consumed 1.071s CPU time.
Jan 23 21:00:00 photobooth systemd[1]: Started go2rtc.service - go2rtc streaming software.Jan 23 21:00:00 photobooth go2rtc[3072]: 21:00:00.327 INF go2rtc version 1.8.6-4 linux/arm64
Jan 23 21:00:00 photobooth go2rtc[3072]: 21:00:00.330 INF [rtsp] listen addr=:8554
Jan 23 21:00:00 photobooth go2rtc[3072]: 21:00:00.330 INF [api] listen addr=:1984
Jan 23 21:00:00 photobooth go2rtc[3072]: 21:00:00.330 INF [webrtc] listen addr=:8555
Jan 23 21:00:00 photobooth go2rtc[3072]: 21:00:00.330 DBG [hass] can't import config: open .storage/core.config_entries: no such file or directory
Jan 23 21:00:10 photobooth go2rtc[3080]: Capturing preview frames as movie to 'stdout'. Press Ctrl-C to abort.
Jan 23 21:00:13 photobooth go2rtc[3080]: ERROR: Movie capture error... Exiting.
Jan 23 21:00:13 photobooth go2rtc[3080]: Movie capture finished (0 frames)
Jan 23 21:00:15 photobooth go2rtc[3072]: 21:00:15.057 DBG [streams] can't stop none producer
Jan 23 21:00:15 photobooth go2rtc[3072]: 21:00:15.057 ERR github.com/dadav/go2rtc/internal/mjpeg/init.go:166 > error="streams: EOF"
Jan 23 21:00:19 photobooth go2rtc[3085]: Capturing preview frames as movie to 'stdout'. Press Ctrl-C to abort.
Jan 23 21:00:20 photobooth go2rtc[3072]: 21:00:20.249 DBG [streams] start producer url=exec:gphoto2 --capture-movie --stdout#killsignal=sigint
Jan 23 21:00:26 photobooth go2rtc[3072]: 21:00:26.037 WRN [streams] json locked
Jan 23 21:00:26 photobooth go2rtc[3072]: 21:00:26.037 DBG [streams] stop producer url=exec:gphoto2 --capture-movie --stdout#killsignal=sigint
Jan 23 21:00:48 photobooth go2rtc[3091]: Capturing preview frames as movie to 'stdout'. Press Ctrl-C to abort.
Jan 23 21:00:51 photobooth go2rtc[3091]: ERROR: Movie capture error... Exiting.
Jan 23 21:00:51 photobooth go2rtc[3091]: Movie capture finished (0 frames)
Jan 23 21:00:52 photobooth go2rtc[3072]: 21:00:52.808 DBG [streams] can't stop none producer
Jan 23 21:00:52 photobooth go2rtc[3072]: 21:00:52.808 ERR github.com/dadav/go2rtc/internal/mjpeg/init.go:166 > error="streams: EOF"
```

Taking pictures / refreshing Photobooth doesn't seem to create any entries as of it doesn't even trigger streaming?! All the entries are from manually accessing the url. Some attempt(s) failed (presumably) because I refreshed before the cam was ready again.

DominicWindisch, 2024-01-24:
In an act of desperation, I've completely wiped my raspberry and reinstalled Photobooth from scratch.

I'm using a Raspberry Pi 4 (2 GB) running 6.1.0-rpi7-rpi-v8.

I've installed photobooth using the installer script with options: `sudo bash install-photobooth.sh -username='dwind' -mjpeg -branch="stable4"`

I've answered
- only page? yes
- remote cups? yes
- print drivers? yes
- sync to usb? yes
- virtual webcam? no

As a camera, I'm (currently) using a Canon EOS M3 which is fully supported by gphoto2.

Now, my weird observations regarding live preview streams:
  - Accessing the DSLR stream remotely using http://192.168.0.174:1984/stream.html?src=dslr&mode=mjpeg works fine.
  - The cam isn't properly released after closing the stream (That'll likely be fixed when https://github.com/AlexxIT/go2rtc/pull/871 is merged)
  - Photobooth will under no circumstances display a live stream from its own go2rtc instance. I've even created a second stream using the virtual webcam as ffmpeg::device input. Same result - stream works fine everywhere but in Photobooth.
  - Photobooth will, however, display MJPEG streams from other go2rtc instances just fine during countdown (as mentioned before from a second pi in the network with a different go2rtc instance up and running) 
  - I also cant seem to get any other kind of preview running. The local device preview just does nothing both on an android phone and a windows laptop. It also doesn't ask for any kind of permission. 

I've tried debugging this in /test/preview.php which just shows the same behavior... There's nothing unusal in the JS logs:
```
tools.js?v=4.4.0:14 [ 2024-01-24T11:09:01.518Z ]: Preview: Preview functions available.
tools.js?v=4.4.0:14 [ 2024-01-24T11:09:08.136Z ]: Starting preview...
tools.js?v=4.4.0:14 [ 2024-01-24T11:09:08.137Z ]: Preview: startVideo mode: 4
tools.js?v=4.4.0:21 [ 2024-01-24T11:09:10.201Z ]: Preview: Preview from URL.
``` 

If I'd use an invalid URL, it would at least complain about that in the logger. Using the proper one just results in nothing whatsoever.

----------

Lastly, I've looked at the usage of `background-image=<url>` in /test/preview.php. _It does not work for photobooths go2rtc streams_. Network monitor tells me its returning content type text:

```
General
Request URL: http://192.168.0.174:1984/stream.html?src=dslr&mode=mjpeg
Request Method: GET
Status Code: 200 OK
Remote Address: 192.168.0.174:1984
Referrer Policy: strict-origin-when-cross-origin

Response Headers
HTTP/1.1 200 OK
Accept-Ranges: bytes
Content-Length: 1699
Content-Type: text/html; charset=utf-8
Date: Wed, 24 Jan 2024 11:44:55 GMT

Request Headers
GET /stream.html?src=dslr&mode=mjpeg HTTP/1.1
Accept: image/avif,image/webp,image/apng,image/svg+xml,image/*,*/*;q=0.8
Accept-Encoding: gzip, deflate
Accept-Language: de-DE,de;q=0.9,en-US;q=0.8,en;q=0.7,pl;q=0.6
Connection: keep-alive
Cookie: org.cups.sid=bf982dcd3b8d3cc83e752169c5219fc6; PHPSESSID=is4gp0nq2em3cvsl1fv50hfohr
DNT: 1
Host: 192.168.0.174:1984
Referer: http://192.168.0.174/
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/120.0.0.0 Safari/537.36
```

Contrary, the go2rtc instance of Octoprint returns jpegs and works fine:
```
General
Request URL: http://192.168.0.73/webcam/stream
Request Method: GET
Status Code:  200 OK
Remote Address: 192.168.0.73:80
Referrer Policy: strict-origin-when-cross-origin

Response Headers
HTTP/1.0 200 OK
access-control-allow-origin: *
content-type: multipart/x-mixed-replace;boundary=123456789000000000000987654321

Request Headers
GET /webcam/stream HTTP/1.1
Accept: image/avif,image/webp,image/apng,image/svg+xml,image/*,*/*;q=0.8
Accept-Encoding: gzip, deflate
Accept-Language: de-DE,de;q=0.9,en-US;q=0.8,en;q=0.7,pl;q=0.6
Connection: keep-alive
DNT: 1
Host: 192.168.0.73
Referer: http://192.168.0.174/
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/120.0.0.0 Safari/537.36

```

 All the other viewers accessing Photobooths go2rtc instance successfully use a `<video>` element in one way or another. I'm out of ideas :(

DominicWindisch, 2024-01-24:
> Thanks for your writeup. Did you also set the capture command to capture %s? This will be needed for the capturing to work. 

Capturing works fine with the default command (as long as gphoto2 was closed properly after streaming).

I've added the modifications to go2rtc.yaml - no change.

I can't see anything suspicious in the network dev tools either. Here's what I'm doing:

- Open /test/preview.php -> network monitor shows loading all resources.
- Click "start preview" -> JS will set the streaming URL as background image and CSS initiates the request
- Network monitor shows the streaming URL being accessed once with the request and return headers as shown above.
- Clicking start preview again won't do anything (obviously, the URL is allready set at this point)

I've cranked up go2trc logging to trace. Here's what I'm getting:
```
24.1.2024, 14:02:30 | trace | [api] GET /api/log 192.168.0.181:52304
24.1.2024, 14:02:36 | trace | [api] GET /api/log 192.168.0.181:52304

<< clicking "start preview" => not getting a stream >>

24.1.2024, 14:02:40 | trace | [api] GET /stream.html?src=dslr&mode=mjpeg 192.168.0.181:52304      
24.1.2024, 14:02:41 | trace | [api] GET /api/log 192.168.0.181:52312
24.1.2024, 14:02:45 | trace | [api] GET /api/log 192.168.0.181:52312
24.1.2024, 14:02:50 | trace | [api] GET /api/log 192.168.0.181:52312
24.1.2024, 14:02:55 | trace | [api] GET /api/log 192.168.0.181:52312
24.1.2024, 14:03:00 | trace | [api] GET /api/log 192.168.0.181:52312

<< opening the stream directly in the browser => getting a stream >>

24.1.2024, 14:03:04 | trace | [api] GET /stream.html?src=dslr&mode=mjpeg 192.168.0.181:52312 
24.1.2024, 14:03:04 | trace | [api] GET /video-stream.js 192.168.0.181:52312
24.1.2024, 14:03:04 | trace | [api] GET /video-rtc.js 192.168.0.181:52312
24.1.2024, 14:03:04 | trace | [api] GET /api/ws?src=dslr 192.168.0.181:52348
24.1.2024, 14:03:04 | trace | [api.ws] msg type=mjpeg
24.1.2024, 14:03:04 | trace | [streams] check cons=0 media=video, sendonly, JPEG
24.1.2024, 14:03:05 | trace | [api] GET /api/log 192.168.0.181:52312
24.1.2024, 14:03:05 | trace | [streams] check prod=0 media=video, recvonly, JPEG
24.1.2024, 14:03:05 | trace | [streams] match prod=0 => cons=0
24.1.2024, 14:03:05 | debug | [streams] start producer url=exec:gphoto2 --capture-movie --stdout#killsignal=sigint
24.1.2024, 14:03:10 | trace | [api] GET /api/log 192.168.0.181:52312

<< closing the tab and stopping the stream >>

24.1.2024, 14:03:14 | trace | undefined error=write tcp 192.168.0.174:1984->192.168.0.181:52348: write: broken pipe caller=github.com/dadav/go2rtc/internal/api/ws/ws.go:120     
24.1.2024, 14:03:14 | debug | [streams] stop producer url=exec:gphoto2 --capture-movie --stdout#killsignal=sigint
24.1.2024, 14:03:15 | trace | [api] GET /api/log 192.168.0.181:52312
24.1.2024, 14:03:20 | trace | [api] GET /api/log 192.168.0.181:52312
24.1.2024, 14:03:25 | trace | [api] GET /api/log 192.168.0.181:52312
24.1.2024, 14:03:30 | trace | [api] GET /api/log 192.168.0.181:52312
```

Feels like the request to go2rtc is wrong somehow? Is there some setting as to why CSS may request stuff differently than the JS solutions?

dadav, 2024-01-24:
That are valuable logs, thanks. I think it could be something wrong with the kill signal parsing, I'll dig into it

dadav, 2024-01-24:
I think you've used the wrong url. It's working in your browser because you've used the html version, but you need the mjpeg version.

Try this:

`url("http://192.168.0.174:1989/api/stream.mjpeg?src=dslr")`

In go2rtc you can get this url by clicking on links (on the main page) and then this middle link:

[image]

simoes-mickael, 2024-01-24:
It's work ! Thank you very much !

closed 2024-01-25 by andi34