PhotoboothProject/photobooth #572: [BUG]: Live View Eos 1300D

Tobi392, 2024-01-26:
### Describe the bug

Hi,
Live view with the Canon EOS 1300D is not working.
Take picture command python3 cameracontrol.py --capture-image-and-download %s works fine.
If I'm using the command python3 cameracontrol.py --bsm for live view preview the shutter clicks the whole time but no preview will be displayed. Also no picture will be taken. Seems to be like a loop. The cheese screen is displayed the whole time.

### Steps to reproduce the behavior

[2024-01-26 20:28:05][main][ERROR] error {"error":"File was not created","cmd":"python3 cameracontrol.py --capture-image-and-download /var/www/html/data/tmp/20240126_202801.jpg 2>&1","returnValue":1,"output":["Traceback (most recent call last):","  File \"/var/www/html/api/cameracontrol.py\", line 726, in ","    sys.exit(main())","             ^^^^^^","  File \"/var/www/html/api/cameracontrol.py\", line 691, in main","    cam = CameraControl(args)","          ^^^^^^^^^^^^^^^^^^^","  File \"/var/www/html/api/cameracontrol.py\", line 100, in __init__","    self.connect_to_camera()","  File \"/var/www/html/api/cameracontrol.py\", line 120, in connect_to_camera","    raise CouldNotConnectException()","CouldNotConnectException"]}
[2024-01-26 20:48:26][main][ERROR] error {"error":"File was not created","cmd":"python3 cameracontrol.py --capture-image-and-download /var/www/html/data/tmp/20240126_204822.jpg 2>&1","returnValue":1,"output":["Traceback (most recent call last):","  File \"/var/www/html/api/cameracontrol.py\", line 726, in ","    sys.exit(main())","             ^^^^^^","  File \"/var/www/html/api/cameracontrol.py\", line 691, in main","    cam = CameraControl(args)","          ^^^^^^^^^^^^^^^^^^^","  File \"/var/www/html/api/cameracontrol.py\", line 100, in __init__","    self.connect_to_camera()","  File \"/var/www/html/api/cameracontrol.py\", line 120, in connect_to_camera","    raise CouldNotConnectException()","CouldNotConnectException"]}

################################
Config: ui
----------------
Subconfig: language
Value:     "en"
----------------
Subconfig: folders_lang
Value:     "resources\/lang"
----------------
Subconfig: notification_timeout
Value:     "5"
----------------
Subconfig: show_fork
Value:     false
----------------
Subconfig: skip_welcome
Value:     false
----------------
Subconfig: admin_shortcut
Value:     true
----------------
Subconfig: admin_shortcut_position
Value:     "bottom-right"
----------------
Subconfig: style
Value:     "modern_squared"
----------------
Subconfig: button
Value:     "modern_squared"
----------------
Subconfig: shutter_animation
Value:     false
----------------
Subconfig: shutter_cheese_img
Value:     ""
----------------
Subconfig: result_buttons
Value:     true
----------------
Subconfig: font_size
Value:     "16px"
----------------
Subconfig: decore_lines
Value:     true
----------------
Subconfig: rounded_corners
Value:     false
----------------
Subconfig: branding
Value:     "Photobooth"
----------------
################################
Config: adminpanel
----------------
Subconfig: view
Value:     "expert"
----------------
Subconfig: experimental_settings
[... 1391 lines cut ...]
################################
Config: photobooth
----------------
Subconfig: version
Value:     "4.3.1"
----------------
Subconfig: basePath
Value:     "\/"
----------------""

### Expected behavior

Live view from the camera

### Photobooth Version

Actual Version

### Operating System

Raspberry Pi4

### Did it work before?

No

Styne13, 2024-01-26:
Seems to be similar to #527 

Tried out the steps mentioned there, but also can not get a live view with my Nikon D3100. If I run the command for live preview as www-data in a separat terminal, I am able to get a preview on the home screen, but only with chromium.

For me it looks like it has something to do with the way photobooth starts the script, but had no time yet to investigate further.

andi34, 2024-01-27:
Duplicate of https://github.com/PhotoboothProject/photobooth/issues/527

closed 2024-01-27 by andi34

andi34, 2024-01-27:
There currently seems to be an issue while using the cameracontrol.py we're looking at already.

For now you could setup again by scratch and run the installer via:
```
sudo bash install-photobooth.sh -username='YourUsername' --mjpeg
```
This will install go2rtc and also allows a remote preview. 
Adjust the Capture command:
`capture %`

And Set preview `from URL` as preview and define the URL: `url("http://localhost:1984/api/stream.mjpeg?src=dslr")`

Another problem might be Firefox which isn't related to Photobooth and can't be fixed by us. Firefox-ESR was told to work.

Tobi392, 2024-01-27:
Hi, thanks for your response.
Live view works now fine.
But with only the command "capture %" i'm running into an error, last log is:
[2024-01-27 13:51:05][main][DEBUG] Capture with CMD {"cmd":"capture %","tmpFile":"/var/www/html/data/tmp/20240127_135105.jpg"}
Do i need to add any download command or something else?

andi34, 2024-01-27:
Sorry, Typo

```
capture %s
```