PhotoboothProject/photobooth #585: [BUG]: Print does not start

o3liger, 2024-01-31:
### Describe the bug

In the admin panel I switched on the function to print immediately after I took a picture. For any reason a print job is not be generated.
If I try to print from the gallery I'm able to print. 

### Steps to reproduce the behavior

1. Turn on printing after taking a photo
2. Take a photo

### Expected behavior

Start printing job

### Photobooth Version

4.4.0

### Browser and browser version

Chrome 121.0.6167.85 (Offizieller Build) (x86_64)

### Operating System

MacOS

### Did it work before?

Yes

andi34, 2024-01-31:
Any error visible inside the browser console? (Press F12 to access the console)

andi34, 2024-01-31:
Also __Allow image filter__ option **must** be disabled to use that function.

o3liger, 2024-01-31:
Turning off Allow image filter option was the solution! Thank you so much!!

closed 2024-01-31 by o3liger