PhotoboothProject/photobooth #626: [BUG]: Raspberry PI Camera Preview and Picture

hillerpa, 2024-02-25:
### Describe the bug

With The config shown on the picture the background works fine but the Picture is not taken and throwing an Error - it seems as the Preview does not stop before taking a picture (and I do not know how to stop it and restart it afterwards) - maybe can you give me a hint what I miss and add this to the FAQ Section?

[image]
[image]

### Steps to reproduce the behavior

System:
RPI5 
PI OS Bookworm 64bit 2023-12-05
Photobooth Version: 4.3.1
Raspberry PI Camera Module V3

### Expected behavior

I try to use the PI Camera as Main Foto Camera as well as Preview on Background. 

### Photobooth Version

4.3.1

### Browser and browser version

Latest Firefox on Bookworm 64bit 2023-12-05

### Operating System

Bookworm 64bit 2023-12-05

### Did it work before?

No

### Additional context

Thank you in advance for the help - I'm really struggling here. I also had the case that the preview was running on startup and pictures are taken but then the preview was not restarted...

Best regards Patrick

hillerpa, 2024-02-25:
it seems as https://photobooth-app.org/setup/shareservice/ is solving this problem - coming back if not

closed 2024-02-25 by hillerpa

STarfield42, 2024-03-03:
@hillerpa : I don't understand : you switched from PhotoboothProject to photobooth-app to fix the problem ?

hillerpa, 2024-03-04:
yes exactly, I did not got it working with PhotoboothProject and photobooth-app worked directly

STarfield42, 2024-03-05:
OK, so may be we can keep the ticket opened as it would be a very interesting feature for Photobooth project ?

andi34, 2024-03-05:
It's more like an issues of Pi foundation / PiOS. As mentioned in the release notes, Firefox should be able to detect the camera as webcam: https://www.raspberrypi.com/news/bookworm-the-new-version-of-raspberry-pi-os/

Sadly it seems not to be true at the moment and it's not an issue of Photobooth.

andi34, 2024-03-06:
You can try running the following script with sudo

https://gist.github.com/andi34/5241ec46ab529c0720d0d65c0de36c18

Note about the Photobooth config Info at the end of the file. Let me know if it works for you.

andi34, 2024-03-06:
```sh
wget https://gist.githubusercontent.com/andi34/5241ec46ab529c0720d0d65c0de36c18/raw/a342fde78716adcf0c23f75b88dec87a4ae2edf5/pi-go2rtc.sh

sudo bash pi-go2rtc.sh
```

andi34, 2024-03-06:
@hillerpa you config **is wrong in general** which might be the cause of problem for you. You're mixing different preview options/preview devices.

cameracontrol.py is **only** used for capture and preview **on devices supported by gphoto2/libgphoto2**!

For Pi Camera it's preview from URL without a command to start a preview.
From the release notes of Pi OS bookworm only Firefox detects Pi Camera as webcam.
This also means: 
- the display **must** be connected to the raspberry
- Photobooth **must** be accessed via http://localhost (using the IP address might not work as it's not treated as safe origin by default)

But: there might be a general issue on PiOS or Firefox at the moment.

The workaround using go2rtc should solve it, but it's blind written ATM and might need to be adjusted.
Go2rtc will also give the possibility to use a different device/display as the stream can be accessed also via IP-Address for a remote preview from URL.

Waiting for feedback on go2rtc as I am currently not able to test myself.

STarfield42, 2024-03-08:
Thank you @andi34 ,

I'll try your script & settings this week end if possible.

loranger, 2024-10-12:
@andi34 

Hi, I had an issue with your gist.
When ran, it failed with a `local: can only be used in a function` error

I replaced `local` with `declare` on lines 8-11 and was able to run your gist

Unfortunately, it does not work, go2rtc is running, but I can't get any preview.
If I try to reach http://localhost:1984/api/stream.mjpeg?src=dslr I get a `stream not found` error

andi34, 2024-10-12:
See https://photoboothproject.github.io/faq/#preview-from-url-remote-preview-preferred and https://photoboothproject.github.io/faq/#setting-up-a-preview-stream-from-your-dslr-or-picamera

Might need go2rtc v1.9.2