PhotoboothProject/photobooth #642: [BUG]: Image frame not working

ZaRdoOx, 2024-03-20:
### Describe the bug

when i select take picture with frame it automatically reboot the photobooth but the path of the frame is correct and im using the release 4.4.0

### Steps to reproduce the behavior

1. click take pic with frame
2. take picture
3. photobooth error

### Expected behavior

when i select take picture with frame it automatically reboot the photobooth but the path of the frame is correct and im using the release 4.4.0

### Photobooth Version

4.4.0

### Browser and browser version

chrome and firefox

### Operating System

windows 11

### Did it work before?

No

andi34, 2024-03-20:
More information about your setup needed.
OS Photobooth installed on?
PHP version?

ZaRdoOx, 2024-03-21:
> More information about your setup needed. OS Photobooth installed on? PHP version?

Hello ! 

PHP Version 8.2.13

OS : Windows 11

andi34, 2024-07-14:
Still need to look at. Busy in real life currently.

sibr1234, 2024-08-25:
For me adding image frames doesn't work either. In the log i found the following:
Error ["Can't create GD resource.","Failed to create frame from image."]
This leads me to image.php and the function call imagcreatetofromstring(). Unfortunately i never did anything with php until now and don't know how to debug...
Thanks for helping

andi34, 2024-08-26:
How does your config look like?
Until v4.4.1 frame needed to be a full path while on latest development version it must be placed inside the Webserver.
Example until v4.4.1
```
/var/www/html/resources/img/frames/frame.png
```
Example on latest development version:
```
/resources/img/frames/frame.png
```

andi34, 2024-08-26:
And if it's failing with `Can't create GD resource` there might be something wrong with your image file.

andi34, 2024-08-26:
[image]

At least on Linux I also can't confirm a bug on v4.4 0/v4.4.1

sibr1234, 2024-08-26:
I am running on RPI5 with 8 Gb RAM. Installed using the install-photobooth script (version 4.99). PHP version is 8.3.10.
I tried it with the demo image and one of the provided frames in the webserver directory

andi34, 2024-08-26:
[image]
[image]

PHP 8.2, frame with and without extension

andi34, 2024-08-26:
Same with php 8.3.10
[image]

sibr1234, 2024-08-27:
I did a clean reinstall and now it works perfectly. Unfortunately i cannot replicate what the problem was.
 Thanks for your help! Appreciate it a lot!

closed 2024-09-28 by andi34