PhotoboothProject/photobooth #648: [BUG]:  picam(hq) stops working when installing photobooth

kugelzucker, 2024-03-30:
### Describe the bug

installed raspberry pi os (latest version as of today) and checked if the picamHQ worked as expected via libcamera-still, which it did.

updated via apt and finally installed photobooth via the shell script.

### Steps to reproduce the behavior

to my surprise after the installation finished and i rebooted i cant get an image via `libcamera-still
`
`ERROR: *** no cameras available ***`

or `gphoto2 --capture-image`
`Could not detect any camera
*** Error (-105: Unknown model)`

### Expected behavior

working picam via libcamera and/or gphoto2

### Photobooth Version

latest via installscript (v4 i guess)

### Did it work before?

Yes

### Additional context

i dont know if it worked before as intended, i just found this project.

kugelzucker, 2024-03-30:
i forgot to add: i flashed a new sdcard to confirm the camera was working as it was sitting there with the cabling the way it was when it didnt work with the sdcard that had photobooth on it. it worked :-/ i am confused.

i tried the usual things like killing the gvfs process and so on.

andi34, 2024-03-31:
Blacklisted while by accident installing gphoto2 preview?
```
sudo rm /etc/modprobe.d/v4l2loopback.conf
sudo rmmod v4l2loopback
sudo reboot
```

andi34, 2024-03-31:
Pi cameras aren't supported by gphoto2, no need to care about gphoto2.

kugelzucker, 2024-03-31:
> Blacklisted while by accident installing gphoto2 preview?
> 
> ```
> sudo rm /etc/modprobe.d/v4l2loopback.conf
> sudo rmmod v4l2loopback
> sudo reboot
> ```

that worked like a charm! you made my day and/or party.

any way this can get integrated in the installer? it was completely vanilla and happened two times so it isnt some kind of fluke.

thanks again for the great project, will donate to the seals!

closed 2024-03-31 by kugelzucker

andi34, 2024-03-31:
```
info "Do you want to install a service to be able to stream your camera to?"
    info "Your camera must be supported by gphoto2 for liveview."
    info ""
    echo "Your options are:"
    echo "1 Install gphoto2 webcam service"
    echo "2 Install go2rtc and needed service to stream your camera via http"
    echo "3 Don't install a service to set up preview for gphoto2"
    info ""
    ask_yes_no "Please enter your choice:" "3"
    info ""
    if [[ $REPLY =~ ^[1]$ ]]; then
        GPHOTO_PREVIEW=true
        MJPEG_PREVIEW=false
        info "### We will install a service to set up a virtual webcam for gphoto2."
        warn "### Note: This will disable other webcam interfaces on a Raspberry Pi (e.g. Pi Camera)."
    elif [[ $REPLY =~ ^[2]$ ]]; then
        GPHOTO_PREVIEW=false
        MJPEG_PREVIEW=true
        info "### We will install a service to set up a mjpeg stream for gphoto2."
    else
        GPHOTO_PREVIEW=false
        MJPEG_PREVIEW=false
        info "### We won't install a service to set up preview for gphoto2."
    fi
```

You choosed `1 Install gphoto2 webcam service, while ` 3 Don't install a service to set up preview for gphoto2`  would have been your option. The webcam service blacklists Pi Camera Kernel module.

kugelzucker, 2024-03-31:
got it, my bad!