PhotoboothProject/photobooth #706: Fatal Error

thetechnikfreak, 2024-05-16:
### Describe the bug

I cant Open the Photobooth: Fatal error: Uncaught Exception: Database or
path is not writable: /var/www/html/data/mail-
adresses.txt in
/var/www/html/src/Service/MailService.php:12
Stack trace: #0 /var/www/html/lib/boot.php(76)
Photobooth\Service\MailService->__construct()
#1 /var/www/html/index.php(3): require_once('...
#2 /main, thrown in
/var/www/html/src/Service/MailService.php on Line 12
[image]

### Steps to reproduce the behavior

Do a fresh install of the Photobooth 

### Expected behavior

IT to work

### Photobooth Version

Newest

### Browser and browser version

Chromium newest

### Operating System

Raspi os

### Did it work before?

Yes

### Latest working version

Before fresh install

andi34, 2024-05-16:
You've used the installer?
The installer should set the permissions right by default.

fmiccolis, 2024-05-16:
I have the same problem on my machines (Windows 10 and 11) both initialized on WSL2 following the guide on [here](https://photoboothproject.github.io/install/install-windows/).

Could be useful to know that also the capture command does not work:
[image]

thetechnikfreak, 2024-05-16:
> You've used the installer?
> The installer should set the permissions right by default.

Yes in tried IT with the Installer wich Not worked and then deleted IT and did It with the Manual Check on the Guide on GitHub.io wich didn't work ether

thetechnikfreak, 2024-05-16:
Plus Informationen: Im on a raspi 4 and i Open the Admin and the Home Page and IT didn't work

andi34, 2024-05-17:
```sh
sudo chown -R www-data:www-data /var/www/html
```
Guess You've missed to set permissions for the www-data user

andi34, 2024-05-17:
Might be related to https://github.com/PhotoboothProject/photobooth/pull/704 , looks like folders are not created ATM

thetechnikfreak, 2024-05-17:
So i Reinstall IT witz the new Version via the Install Script but i Had still to run the permision command

PR #708 (2024-05-19): bugfix: create folders as needed
Fix #706
closed 2024-05-17 by thetechnikfreak