PhotoboothProject/photobooth #712: [BUG]: capture error

thetechnikfreak, 2024-05-18:
### Describe the bug

Here is the next problem : curl --data "style=photo" http://192.168.178.71/api/capture.php  Warning:  Undefined array key "canvasimg" in /var/www/html/api/capture.php on line 84  Deprecated:  explode(): Passing null to parameter #2 ($string) of type string is deprecated in /var/www/html/src/PhotoboothCapture.php on line 62  Warning:  Undefined array key 1 in /var/www/html/src/PhotoboothCapture.php on line 62  Deprecated:  explode(): Passing null to parameter #2 ($string) of type string is deprecated in /var/www/html/src/PhotoboothCapture.php on line 63  Warning:  Undefined array key 1 in /var/www/html/src/PhotoboothCapture.php on line 63  Deprecated:  base64_decode(): Passing null to parameter #1 ($string) of type string is deprecated in /var/www/html/src/PhotoboothCapture.php on line 64  Warning:  file_put_contents(/var/www/html/data/tmp/20240518_064852.jpg): Failed to open stream: No such file or directory in /var/www/html/src/PhotoboothCapture.php on line 66 {"success":"image","file":"20240518_064852.jpg"}

### Steps to reproduce the behavior

Make a photo

### Expected behavior

Make a photo

### Photobooth Version

Newest

### Browser and browser version

Chromium newest

### Operating System

Raspi os

### Did it work before?

No

thetechnikfreak, 2024-05-18:
Update: After i rebuild the Image Database IT says Error saving Image Warnung failed to create Thumbnail

andi34, 2024-05-19:
Since you're using curl instead of the regular capture function I can not confirm the issue and the issue should be related to your setup and missing send data properties.

To create proper bug reports please use the capture function via start page. Make sure Loglevel is set to 2. Debugpanel is available via admin panel and has access to the Photobooth log.

closed 2024-05-19 by andi34

andi34, 2024-05-19:
To avoid your issue because of wrong configuration and own usage of the API https://github.com/PhotoboothProject/photobooth/pull/713

thetechnikfreak, 2024-05-19:
@andi34 There was the Same issue via the web in the log

thetechnikfreak, 2024-05-19:
[image]

andi34, 2024-05-19:
That's a different error and fixed already by https://github.com/PhotoboothProject/photobooth/pull/708

commit 01cda0b5 (2024-05-19): capture: throw error on missing canvas data (#713)

Fix #712