PhotoboothProject/photobooth #749: [BUG]: Connected network mandatory for hardware Button (RPI)

e0601180, 2024-06-12:
### Describe the bug

Hello,

I have installed the software on a Raspberry PI 4 (with Raspberry Pi OS Legacy (Debian bullseye)), and configured a hardware button using GPIO.

Unfortunately, I observed that this button doesn't work at all is there is no network connected ..
Is it a normal usage or a bug ?

That's a big problem for me because I will use this photobooth for a wedding in a place where no WLAN/LAN is available.

Thanks for your answers.

### Steps to reproduce the behavior

1. Install photobooth software on a raspberry pi 4 (RPI OS Legacy)
2. Connect an hardware button on GPIO pin
3. Check that no network is connected 
4. Launch photobooth software
5. Try to use the hardware button

### Expected behavior

Hardware button is working even if there is no network connected

### Photobooth Version

4.99

### Did it work before?

No

andi34, 2024-06-12:
6.6 Kernel is not supported. 

Independent of Photobooth:
It might be, that the Remotebuzzer server (socket.io) requires an network connection. You can use any router you have and connect the Pi to it (even if it isn't connected to the Internet). Personally I am using a travel router and allow download via QR Code.

e0601180, 2024-06-12:
Thanks for your answer.

As I mentionned, I am using Raspberrypi OS Legacy (Debian11 (bullseye)), so kernel version is 6.1.

Ok, I understand that this is a "normal" behaviour. I will use a share connection with a cellphone if necessary.

Is there any way to make it work "offline" ? If not, can it be a new feature request ?

andi34, 2024-06-12:
I'd say it's independent of Photbooth and therefore no feature request needed and that's also why the bug was closed.
https://github.com/socketio/socket.io/issues/2309

e0601180, 2024-07-04:
Hello,

Sorry to open this issue again.

I tried yesterday to share a network connection with my smartphone and connect my RPI to it.
But result is the same, it's not working. In the logs, there is no line like "New client connected - ID xxx".
I don't know if it's a normal behaviour.

Anyway, I would like to test with a proper router.
You mentionned a travel router, can you share the model ?
I found this one : TP-Link TL-WR902AC, but I am not sure if it can make it.

Also, how should I configure the router and the software to make it work together?

Thx

andi34, 2024-07-04:
Make sure the IP for the Remotebuzzer server is set correctly. If it was set once with a different IP you need to adjust it.

For WLAN connected devices make sure they are allowed to communicate via network, on a FritzBox this is an own setting which needs to be enabled.

If you're having further trouble make sure setting the Loglevel 2 and check the Remotebuzzer log, also the browser log.