PhotoboothProject/photobooth #797: [BUG]: Missing a description what Photobooth is and what not

Mannshoch, 2024-07-17:
### Describe the bug

I search for a Image server software where I could stor our families Photos. From the description of the Feature It is not so clear what photobooth is. I currently only assume that it is a software for manage photos.

### Steps to reproduce the behavior

1. Assume you do not know what fotoboot is.
2. Open Github and try to find a text that describe the usecase on the readme.
3. go to https://photoboothproject.github.io
4. read the FAQ
5. Currently, I assume it is a Picture server with the possibility to add a camera.

### Expected behavior

I expected a text that tell what this software is for, may what not.

### Photobooth Version

-

### Did it work before?

Yes

andi34, 2024-07-17:
Hey and thanks for your recommendation.
I guess the name Photobooth describes well it's not an gallery only. https://en.m.wikipedia.org/wiki/Photo_booth

The web page is deployed from the files inside the docs/ folder of the source. You're gladly welcome to contribute to the project and webpage and create a pull request.

Best regards 

Andi

Mannshoch, 2024-07-17:
I'm not an English speaker. I had to check what booth means on dictionary. At first I translated it with a storage room or kind of a convenient shop.  I didn't had the idea to check Wikipedia.
(for the record, currently I evaluate piwigo)

closed 2025-11-28 by andi34