PhotoboothProject/photobooth #802: [BUG]: Stable 4 installation fails on Debian 12

2302Dave, 2024-07-20:
### Describe the bug

Want to install Photobooth Stable 4 on Debian 12 fresh installation. Fails with error.

`[...]
node-deep-equal (2.1.0+~cs31.12.80-1) wird eingerichtet ...
npm (9.2.0~ds1-1) wird eingerichtet ...
Trigger für man-db (2.11.2-2) werden verarbeitet ...
[Info]      Found npm 9.2.0
[WARN]      npm needs to be updated!

added 1 package in 8s

27 packages are looking for funding
  run `npm fund` for details
[Info]      npm available..
[Info]      Found npm 9.2.0
[ERROR]     Update of npm was not possible. Aborting Photobooth installation!
root@photobooth:~/photobooth# npm fund
photobooth@4.4.1`

### Steps to reproduce the behavior

1. Install Debian 12
2. Clone Repo - # git clone https://github.com/PhotoboothProject/photobooth.git
3. Checkout Stable 4 branch - # git checkout stable4
4. Start installation - # bash install-photobooth.sh -username='photobooth'
5. Fails with error

`[...]
node-deep-equal (2.1.0+~cs31.12.80-1) wird eingerichtet ...
npm (9.2.0~ds1-1) wird eingerichtet ...
Trigger für man-db (2.11.2-2) werden verarbeitet ...
[Info]      Found npm 9.2.0
[WARN]      npm needs to be updated!

added 1 package in 8s

27 packages are looking for funding
  run `npm fund` for details
[Info]      npm available..
[Info]      Found npm 9.2.0
[ERROR]     Update of npm was not possible. Aborting Photobooth installation!
root@photobooth:~/photobooth# npm fund
photobooth@4.4.1`

### Expected behavior

Install should properly finish.

### Photobooth Version

4.4.1

### Operating System

Linux photobooth 6.1.0-23-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.99-1 (2024-07-15) x86_64 GNU/Linux /// Debian GNU/Linux 12

### Did it work before?

No

### Additional context

How to install stable version? Tried dev branch, which installs, but can't take photos from Canon EOS 600D. Stable 4.4.1 fails on installation. Documentation seems to mention only dev branch installation.

andi34, 2024-07-20:
If you can't capture from your camera it's not related to the Photobooth version.

Update npm by hand if needed, also make sure using the installer from dev branch and define the branch if needed.

```
wget -O install-photobooth.sh https://raw.githubusercontent.com/PhotoboothProject/photobooth/dev/install-photobooth.sh
sudo bash install-photobooth.sh -username='<YourUsername>' -branch="stable4"
```

Note:
Stable4 is deprecated, and won't be updated, errors won't be looked at anymore.

andi34, 2024-07-20:
For reference;
https://photoboothproject.github.io/faq/#i-get-the-error-message-something-went-wrong-while-taking-a-picure-what-can-i-do

https://photoboothproject.github.io/faq/#gphoto2-troubleshooting

closed 2024-07-20 by andi34