PhotoboothProject/photobooth #819: [BUG]: enable-usb-sync.sh doesn't work with Ubuntu 24.04 LTS

carsten-wolf, 2024-08-07:
### Describe the bug

Because the polkit Local Authority no longer exists at Ubuntu 24.04 LTS the enable-usb-sync.sh failds to add the setting for the www-data user that enables the USB Sync function.

https://manpages.ubuntu.com/manpages/noble/en/man8/pklocalauthority.8.html

I have tried to add the www-data user to the disk group, but still getting error messages from the DEBUG Admin Console that writing is not possible.

### Steps to reproduce the behavior

Use Ubuntu oder XUbuntu 24.04 LTS and execute the enable-usb-sync.sh script

### Expected behavior

Error message that the 50-local.d folder is not existing

### Photobooth Version

4.4.1

### Operating System

XUbuntu 24.04 LTE

### Did it work before?

No

andi34, 2024-08-09:
Please try the following:
```sh
sudo nano etc/polkit-1/localauthority.conf.d/10-photobooth.pkla
```

Add the following content and save the file. After that reboot once please:
```
[Allow www-data to mount drives with udisks2]
Identity=unix-user:www-data
Action=org.freedesktop.udisks2.filesystem-mount*;org.freedesktop.udisks2.filesystem-unmount*
ResultAny=yes
ResultInactive=yes
ResultActive=yes
```

I guess only the location changed from

/etc/polkit-1/localauthority/50-local.d/

to

/etc/polkit-1/localauthority.conf.d/

carsten-wolf, 2024-08-10:
Thanks for the fast feedback. I'll take a look an try it.

carsten-wolf, 2024-08-20:
It didn't worked for me.

Because of a lot of other Problems I decided to do a reinstall with XUbuntu 22.04. That is much better.
Also I installed the photobooth html direct in the /var/www and not in /var/www/photobooth, because for example the fonts and also the cheese images didn't work. I guess there is a problem if with the link to the resources folder if it's not installed in the root.

andi34, 2024-08-20:
Guess you had a wrong server configuration or wrong path. Default install paths are `/var/www/html` or `/var/www/html/photobooth`. For some functions Photobooth must be installed inside the Webservers root folder / subfolder of the root folder and won't work if they are placed outside the Webserver.

For the usb-sync:
You must also prevent the system from auto mounting, ATM this can only be setup on Pi OS. The Webserver is not able to unmount or mount if you logged in user has mounted the USB drive.

closed 2024-10-27 by carsten-wolf