PhotoboothProject/photobooth #822: [BUG]: QR Code always put _/ in front of the URL (QR-Code scanned with pixel 8 pro)

BenZeit, 2024-08-09:
### Describe the bug

When I scan the QR-Code to download the picture, it always put a "_/" in front of the URL.
It looks like: "_/192.168.0.105/api/download.php?image=filename.jpg"
the config in the admin-panel: 192.168.0.105/api/download.php?image=
I cant understand why it puts that "_/" in front of the URL. Is there a fix for that? What did I configured wrong?

### Steps to reproduce the behavior

1. Make a picture
2. Click on QR-Code
3. Scan QR Code with Pixel 8 Pro
4. geht the URL with _/ in front 

### Expected behavior

get the URL without _/ by scanning the qr code

### Photobooth Version

4.99

### Did it work before?

No

andi34, 2024-08-09:
Could You please post your URL config?

andi34, 2024-08-09:
If it's
```
192.168.0.105/api/download.php?image=
```
You **must** also add the protocol.

```
http://192.168.0.105/api/download.php?image=
```
In that case it's an configuration error of your setup and not a bug.

closed 2024-08-09 by andi34

BenZeit, 2024-08-09:
> If it's
> 
> ```
> 192.168.0.105/api/download.php?image=
> ```
> 
> You **must** also add the protocol.
> 
> ```
> http://192.168.0.105/api/download.php?image=
> ```
> 
> In that case it's an configuration error of your setup and not a bug.

That was the solution. Sorry for that and big thanks for the fast reply

MKanister, 2025-07-23:
Hello, wouldn't this solution be something for the FAQ, since there's nothing there about the syntax - it helped me a lot.

andi34, 2025-07-24:
You're welcome to add it and create a Pull request.
Source of the FAQ comeswith Photobooth https://github.com/PhotoboothProject/photobooth/blob/dev/docs/faq/index.md